microsoft/AirSim #2569: Build error(ros package)

bjzenglei, 2020-04-13:
When I Compile with catkin_make under Airsim/ros，it occured a error（c++: error: unrecognized command line option ‘-stdlib=libc++’）。I have run ./setup.sh and ./build.sh succeed。
OS: Ubuntu 16.04 LTS
AirSim: Master
Python: python3.5
UnReal: UE4.22
--------------------------------------------------------------------------------------------------------------------------------------------------
[  0%] Built target geographic_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  1%] Building CXX object airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/home/helius/UE4/AirSim_tmp/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o
c++: error: unrecognized command line option ‘-stdlib=libc++’
airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:62: recipe for target 'airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/home/helius/UE4/AirSim_tmp/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o' failed
make[2]: *** [airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/home/helius/UE4/AirSim_tmp/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o] Error 1
CMakeFiles/Makefile2:3899: recipe for target 'airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/all' failed
make[1]: *** [airsim_ros_pkgs/MavLinkCom/CMakeFiles/MavLinkCom.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_LandGroup
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Land
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleQuatCmd
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_VelCmd
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_SetGPSPosition
--------------------------------------------------------------------------------------------------------------------------------------------------

rajat2004, 2020-04-13:
This problem is being discussed at https://github.com/microsoft/AirSim/issues/2551, will be better to have a single issue than duplicate ones

bjzenglei, 2020-04-13:
> This problem is being discussed at #2551, will be better to have a single issue than duplicate ones

Thanks very much

madratman, 2020-04-13:
Yes, let's track in #2551. I just opened PR #2571 to fix this.

closed 2020-04-13 by madratman