microsoft/AirSim #2570: Flying straight while taking pictures

alonyomtov123, 2020-04-13:
I am trying to fly a quadcopter wihle taking pictures.
I'm having 2 problems:
The first is that the drone won't stop in place so I can take the picture properly.
The second is that the flight path is very windy. The drone doesn't fly straight from one dot to another.

I am using Ubuntu 16.04 with python 3.7.

rajat2004, 2020-04-14:
Well, a drone doesn't exactly behave like you want. Maybe try Computer Vision mode

alonyomtov123, 2020-04-15:
I expected it not to fly exactly straight but it moves up and down and the cameras take pictures which are not realistic. Is there a way to make the drone not move between movement commands or while taking a picture?

cloudlakecho, 2020-05-21:
@alonyomtov123, I have the same challenge with you. What I found at an example of `PythonClient\multirotor\orbit.py` is that I need to stop the drone and take a picture.

closed 2020-08-17 by jonyMarino