microsoft/AirSim #2573: Issue with PX4

johan606303, 2020-04-13:
Hi.

I want to use the controller PX4 for my drones, in Software-in-Loop mode.

So as I use windows 10, I install first Cygwin Toolchain.

Then I open it's console (run-console.bat) run the following commands : 

mkdir -p PX4
cd PX4
git clone https://github.com/PX4/Firmware.git
cd Firmware
git checkout v1.9.2

but when I run : 
make px4_sitl_default none_iris

I have the following errors : 
[image]

Thank you for help.

bys1123, 2020-04-13:
Cygwin is using gcc 9, so will have few errors on 1.9.2. Suggest try 1.10.2.

johan606303, 2020-04-14:
I don't see any difference : 

I still have the following error : 

[0/1] Re-running CMake...
-- PX4 version: v1.10.2
-- PX4 config file: /cygdrive/d/PX4/home/Firmware/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-64-gdec1ed0
-- Configuring done
-- Generating done
-- Build files have been written to: /cygdrive/d/PX4/home/Firmware/Firmware/build/px4_sitl_default
[5/528] Building C object src/lib/mixer/CMakeFiles/mixer.dir/mixer_load.c.o
FAILED: src/lib/mixer/CMakeFiles/mixer.dir/mixer_load.c.o
/usr/bin/cc -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -D_GNU_SOURCE -D__CUSTOM_FILE_IO__ -D__PX4_CYGWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -D__USE_LINUX_IOCTL_DEFS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -Isrc/lib/mixer -O2 -g -DNDEBUG   -U__CUSTOM_FILE_IO__ -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fno-common -Wbad-function-cast -Wnested-externs -Wstrict-prototypes -std=gnu11 -MD -MT src/lib/mixer/CMakeFiles/mixer.dir/mixer_load.c.o -MF src/lib/mixer/CMakeFiles/mixer.dir/mixer_load.c.o.d -o src/lib/mixer/CMakeFiles/mixer.dir/mixer_load.c.o   -c ../../src/lib/mixer/mixer_load.c
In file included from ../../src/lib/mixer/mixer_load.c:42:
../../src/lib/mixer/mixer_load.c: In function ‘load_mixer_file’:
../../src/lib/mixer/mixer_load.c:72:42: error: array subscript has type ‘char’ [-Werror=char-subscripts]
   72 |   if ((strlen(line) < 2) || !isupper(line[0]) || (line[1] != ':')) {
      |                                      ~~~~^~~
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
[18/528] Building C object platforms/commo...e/CMakeFiles/work_queue.dir/hrt_thread.c.
ninja: build stopped: subcommand failed.
make: *** [Makefile:195: px4_sitl_default] Error 1

johan606303, 2020-04-21:
I had better results with : git checkout v1.11.0-beta1. Perhaps it is it.

ahmed-elsaharti, 2020-07-25:
Had this same issue with v1.10.1 (from the docs).
Solved by installing Cygwin Toolchain 0.8 rather than 0.9

jonyMarino, 2020-08-28:
This is a PX4 issue, not an airsim issue

closed 2020-08-28 by jonyMarino