microsoft/AirSim #2575: PX4 HIL OriginGeopoint Issue (Airsim 1.22)

731avenger, 2020-04-14:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
I am attempting to run what seems like a very simple scenario. I need define Airsim the included PX4 HIL sim with a real origin/home position. After reading through the documentation their is two approaches one can take:
One is to use the params setting which define the home position in the SIL sim. https://microsoft.github.io/AirSim/px4_sitl/#setting-gps-origin
```
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "VehicleSysID": 3,
            "SerialPort": "COM16",
            "UseSerial": true,
            "params": {
               "LPE_LAT": 33.33313889,
               "LPE_LON": -84.2666667
                }
        }
    }
```
However, this approach seems to lead to very mixed results with the PX4 origin being set all over the globe rather inconstantly. In the SIL sim, this method works fine but as soon as you introduced the HIL sim, this method seems to no longer work on PX4 1.10.1 running on the Mini Pixhawk 4 (FMU5).

The second approach is to use the OrginGeopoint option. 
```
	"OriginGeopoint": {
        "Latitude": 33.33313889,
        "Longitude": -84.2666667,
        "Altitude": 20
    },
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "VehicleSysID": 3,
            "SerialPort": "COM16",
            "UseSerial": true,
			"Z": 180
        }
    }
```
This method at least got me in the right zip code. However, when I define "33.33313889, -84.2666667" I expect to see that in Qground Control, but QGC is showing a Lat of 33.3487423 and a long -84.240305 which is ~4Km to the northeast of my desired point. 

I am running the pre-built plains map on Airsim 1.22. I have attempted this with a number of pixhawk 4s and Mini 4s running PX4 1.10, 1.10.1, 1.9 and 1.11. I am running Airsim on Windows 10 64-bit with the latest  updates.

rajat2004, 2020-04-14:
The `params` section in `settings.json` was added after 1.2.2 release, and therefore doesn't have any effect on the Home position
As to the difference in the positions from `OriginGeoPoint`, I think the PlayerStart in the env is causing the change. The PlayerStart in UE also has its own offset, which is being added to the Origin set in `settings.json`. Don't think this can be changed since it's a binary, but an easy fix could be to observe the difference and then compensate for that in `settings.json` to get the desired coordinates

731avenger, 2020-04-14:
@rajat2004 I like your idea of just computing the offset, it makes sense that your updating the "0,0" part of the environment rather then player position. I assume the new `param` section will update the player's position when using the PX4 rather then updating the world position like `OriginGeopoint` ? 

As you suggested I calculated my offset, plugged those numbers in to my `settings.json` 
Desired Position:
```
        "Latitude": 33.33313889,
        "Longitude": -84.2666667,
```

New Calculated Offset:
```
         "Latitude": 33.31753548,
        "Longitude": -84.2929529,
```

New PX4 HIL GPS location:
```
Lat: 33.331389
Lon: -84.2666714
```
I am now within about 600m of my desired location, but no matter how much I correct for the offset I still can not get my vehicle in the desired location. What seem like small corrections will send the vehicle kilometers away while large corrections will make small changes or no change. Would it be wise to use a combination of `OrginGeopoint` settings and local `X` and `Y` positions to achieve the correct location?

rajat2004, 2020-04-14:
You could, but I think it should be doable with just `OriginGeopoint`. From my usage, the X,Y,Z values are generally used to work with multiple drones, where we have a common "origin" and each vehicle at a specific offset from there, such as 3m, 6m, etc.

Maybe I'm not understanding this properly due to being a bit sleepy, but shouldn't just setting `Lat` to `33.31753548` and the corresponding Long value like you calculated work? Why the other values which you've mentioned below that?
My reasoning -  `33.3487423 - 33.33313889 = 0.01560341` => Offset created by the env
therefore, `x + 0.01560341 = 33.33313889`, and x gives the location to be set in OriginGeopoint

MSBGit, 2020-04-14:
I am having the same issue. I am running the latest AirSim (Beta 1.2.0) and Unreal 4.24 and using Pixhawk 4 running PX4 in a HITL environment on Windows 10 and using my own custom Unreal environment. My `settings.json` file specifies this `OriginGeopoint`:

```
  "OriginGeopoint": {
    "Latitude": 37.776580,
    "Longitude": -122.309579,
    "Altitude": 0
  }
```
and I have my _PlayerStart_ Actor at this offset from the origin of my environment:

[image]

When I press _Play_ to start the simulation, my Pixhawk gets GPS coordinates that are offset from where the `OriginGeopoint` is set. You can see this on the QGC screenshot:

[image]

If I move _PlayerStart_ to the origin of my environment (Transform of 0, 0, 0), my Pixhawk gets the correct GPS coordinates when I start the simulation, but the problem is that I want my drone to launch from a specific location in my environment, not from 0,0,0.

When running _Blocks_ in Unreal 4.22 with the older AirSim (from last year), the GPS coordinates in the `OriginGeopoint` would get sent to the Pixhawk correctly and not depend on the offset of where the drone started from.

My understanding of `OriginGeopoint` is that it represents the GPS coordinates that are given to the Pixhawk when the simulation starts and which are relative to the launch point of the drone (i.e. where _PlayerStart_ is placed) and not relative to the origin of the environment.

[settings.json.txt](https://github.com/microsoft/AirSim/files/4477075/settings.json.txt)

731avenger, 2020-04-14:
In my case, I really don't care where the environment is, I just need to define the exact position of the aircraft. I see great potential in the `params` setting for my application because it allows me to set a more generic point for the environment while exactly defining the location of the aircraft.

MSBGit, 2020-04-16:
#2588

asherikov, 2020-07-03:
1. "params" config section was renamed to "Parameters" some time ago, documentation has not been updated. "Parameters" still doesn't seem to work for some other reason.
2. `LPE_*` parameters are for LPE estimator, while you are probably using EKF (the default).
3. The error in position can come from two sources: 
- `OriginGeopoint` is not the original position of the vehicle, see discussion linked above;
- AirSim applies very crude conversions between NED frame used internally and GPS coordinates exposed to PX4.
4. Orientation of the vehicle is crappy too since the magnetic field model is very crude as well.

jonyMarino, 2020-11-19:
#3147

closed 2020-11-19 by jonyMarino