microsoft/AirSim #2578: Can't change image resolution in pre-built binary

ndalton12, 2020-04-14:
Running Ubuntu 18.04, Python 3.8.2

Using the pre-built AirSimNH from [here](https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux), I am running the code from [here](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/multi_agent_drone.py). The code and settings works for the blocks environment using a build on linux to take pictures at the desired 1280 x 720 x 3 resolution. However, when running the code on the AirSimNH binary, the image size is 368640. I get the following error:

`
Traceback (most recent call last):
  File "multi_drone.py", line 93, in <module>
    img_rgb = img1d.reshape(response.height, response.width, 3)  # reshape array to 3 channel image array H X W X 3
ValueError: cannot reshape array of size 3686400 into shape (720,1280,3)
`

My settings.json looks like the following:

> {
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
   "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 1280,
          "Height": 720,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        }
     ]
  },
  "Vehicles": {
        "Drone1": {
            "VehicleType": "SimpleFlight",
            "X": 4, "Y": 0, "Z": -2
        },
        "Drone2": {
            "VehicleType": "SimpleFlight",
            "X": 8, "Y": 0, "Z": -2
        }
   }
}

rajat2004, 2020-04-14:
Any reason not to use `v1.3.0` or later binary? The Neighbourhood binary is there for Linux in `v.1.3.0`

ndalton12, 2020-04-14:
No reason in particular, didn't see that 1.3.0 has Neighbourhood. Just tried it out and looks like the issue is gone now. Thanks!

closed 2020-04-14 by ndalton12

saihv, 2020-04-14:
Just FYI, the previous version was returning a 4 channel image instead of
3, the resolution was correct. 1280x720x4 = 3686400

rajat2004, 2020-04-14:
Ahh yes, thanks for the info @saihv!