microsoft/AirSim #2582: Actor tags not being recognized with simGet|SetObjectPose() Python APIs

MSBGit, 2020-04-14:
I am running AirSIm 1.2.0 Beta in my own custom Unreal environment, running Unreal 4.24 on Windows 10. I am using HITL mode with a Pixhawk 4 running PX4 1.10.1.

I am using the AirSim `simGetObjectPose()` and `simSetObjectPose()` APIs from Python. These work fine if I specify the Actor name, however if I specify a tag that has been configured for the Actor, these APIs report that the object isn't found.

For example, I have a StaticMeshActor named `sm_containership-1`. If I call `simGetObjectPose()` with this name, I get the pose data back:

`pose = client.simGetObjectPose("sm_containership-1")`

If I configure one or more _Component Tags_ for this Actor and then use those tags like this:

`pose = client.simGetObjectPose("ship-1")`

 I get back `NaN` (which means the object wasn't found). Here is how I have my tags configured:

[image]

The [AirSIm APIs->Common APIs](https://microsoft.github.io/AirSim/apis/#common-apis) docs say that tags added to an Actor can be used for these APIs.

seanmcleod, 2020-04-15:
@MSBGit any reason that you have defined 2 tags for `sm_containership-1`, i.e. `ship-1` and `ship-2`. Maybe it's only comparing against the 2nd tag so `ship-1` doesn't find any matches?

MSBGit, 2020-04-15:
@seanmcleod my screenshot is a bit confusing. I first tried with a single tag and that didn't work, then I tried with two tags and neither of those tags worked. This simpler screenshot shows a single tag, and that doesn't work:

[image]

MSBGit, 2020-05-18:
It turns out that tags do work, but it's a bit convoluted in Unreal Editor.

- Define an actor tag by clicking on the main object and scroll to _Actor_ in the details pane
- Click the little arrow that expands the advanced options
- And add an element to the tag array
- Go to _Edit_ -> _Project Settings_ and in the search bar type _tags_
- Find _Add New Gameplay Tag_
- Add your new tag and a comment if you wish and press _Add New Tag_
- AirSim will now see the new tag and you can use it in the APIs that take an actor name or tag