microsoft/AirSim #2587: Invoking "make -j4 -l4" failed

EdwinMeriaux, 2020-04-15:
Hello,

So I am using ubuntu 18.04

on the step catkin_make

I am getting this error:

Base path: /home/edwin-pcp/AirSim/ros
Source space: /home/edwin-pcp/AirSim/ros/src
Build space: /home/edwin-pcp/AirSim/ros/build
Devel space: /home/edwin-pcp/AirSim/ros/devel
Install space: /home/edwin-pcp/AirSim/ros/install
####
#### Running command: "cmake /home/edwin-pcp/AirSim/ros/src -DCATKIN_DEVEL_PREFIX=/home/edwin-pcp/AirSim/ros/devel -DCMAKE_INSTALL_PREFIX=/home/edwin-pcp/AirSim/ros/install -G Unix Makefiles" in "/home/edwin-pcp/AirSim/ros/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/edwin-pcp/AirSim/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/edwin-pcp/AirSim/ros/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.23
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - airsim_ros_pkgs
-- ~~  - airsim_tutorial_pkgs
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'airsim_ros_pkgs'
-- ==> add_subdirectory(airsim_ros_pkgs)
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/
-- found RPC_LIB_INCLUDES= /home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim//external/rpclib/rpclib-2.2.1/include
-- Boost version: 1.65.1
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- airsim_ros_pkgs: 5 messages, 7 services
-- +++ processing catkin package: 'airsim_tutorial_pkgs'
-- ==> add_subdirectory(airsim_tutorial_pkgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
-- Generating done
-- Build files have been written to: /home/edwin-pcp/AirSim/ros/build
####
#### Running command: "make -j4 -l4" in "/home/edwin-pcp/AirSim/ros/build"
####
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_eus
[  1%] Linking CXX static library /home/edwin-pcp/AirSim/ros/devel/lib/libMavLinkCom.a
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GPSYaw
[  1%] Built target geometry_msgs_generate_messages_py
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_VelCmd
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleEulerCmd
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Takeoff
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Land
[... 302 lines cut ...]
         if (std::strftime(str, sizeof(str), format, std::localtime(&tt)))
                                                                        ^
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp: In static member function ‘static common_utils::Utils::string common_utils::Utils::to_string(common_utils::Utils::time_point<std::chrono::_V2::system_clock>, const char*)’:
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp:510:72: warning: format not a string literal, format string not checked [-Wformat-nonliteral]
         if (std::strftime(str, sizeof(str), format, std::localtime(&tt)))
                                                                        ^
[ 93%] Building CXX object airsim_ros_pkgs/AirLib/CMakeFiles/AirLib.dir/home/edwin-pcp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibServer.cpp.o
In file included from /home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/Common.hpp:14:0,
                 from /home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/vehicles/multirotor/api/MultirotorRpcLibServer.hpp:7,
                 from /home/edwin-pcp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibServer.cpp:10:
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp: In static member function ‘static common_utils::Utils::string common_utils::Utils::to_string(time_t, const char*)’:
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp:501:72: warning: format not a string literal, format string not checked [-Wformat-nonliteral]
         if (std::strftime(str, sizeof(str), format, std::localtime(&tt)))
                                                                        ^
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp: In static member function ‘static common_utils::Utils::string common_utils::Utils::to_string(common_utils::Utils::time_point<std::chrono::_V2::system_clock>, const char*)’:
/home/edwin-pcp/AirSim/ros/src/airsim_ros_pkgs/../../../../AirSim/AirLib/include/common/common_utils/Utils.hpp:510:72: warning: format not a string literal, format string not checked [-Wformat-nonliteral]
         if (std::strftime(str, sizeof(str), format, std::localtime(&tt)))
                                                                        ^
[ 93%] Linking CXX static library /home/edwin-pcp/AirSim/ros/devel/lib/libAirLib.a
[ 93%] Built target AirLib
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Did I do something wrong? I apologize I've never seen this error before

rajat2004, 2020-04-16:
GCC 8.0 or later is required to build the ROS package. the github.io page needs to be manually updated which is a big headache right now, and causes it to be slightly behind sometimes
See https://github.com/microsoft/AirSim/blob/master/docs/airsim_ros_pkgs.md

rajat2004, 2020-04-16:
I think the problem has been solved, issue can be closed now

EdwinMeriaux, 2020-04-19:
Yes Sorry.
Thank you for your help!

closed 2020-04-19 by EdwinMeriaux