microsoft/AirSim #2588: Clarification on the OriginGeopoint

courageon, 2020-04-15:
There seems to be conflicting data on what the "OriginGeopoint" is. In the documentation it's referenced as the location of the PlayerStart while in code it's commented as the coordinates of Unreal level at the coordinates 0,0,0.

Code:
https://github.com/microsoft/AirSim/blob/e24a11d5985c1d6f573735f28a3844bc634559db/AirLib/include/common/AirSimSettings.hpp#L350

Documentation:
https://microsoft.github.io/AirSim/settings/#origingeopoint

Which leads into my larger question of what would need to be modified in-code, or otherwise to correspond an arbitrary point in Unreal to a specified lat-lon-alt?

Thanks!

MSBGit, 2020-04-16:
This looks like it's related to: [PX4 HIL OriginGeopoint Issue (Airsim 1.2.2) #2575](https://github.com/microsoft/AirSim/issues/2575).

I can see that it would be useful to be able to specify the GPS coordinates of either the origin of the Unreal level at (0,0,0) _and_/_or_ of the _PlayerStart_ Actor. Or maybe it would be even more useful to be able to place a special actor called something like _OriginGeopoint_ in the level which would then use the `OriginGeopoint` GPS coordinates in the `settings.json` file and all other objects int he level would be referenced to that. That would allow the simulation to use whatever origin point makes the most sense for the work being done. So if someone wanted to have  _PlayerStart_ (and hence the drone initial position) be at the GPS coordinates specified in `OriginGeopoint`, they could put both Actors at the same location in the level and get that desired behavior. Or if you had some specific real world object in your world that you wanted to bind to a specific real world GPS coordinate, then you would put your _OriginGeopoint_ Actor at the location of that real word object.

courageon, 2020-04-16:
I really like the actor idea! Something that you can place and then specify the lat/lon/alt in the editor? Otherwise if the actor is not present we can default to the current behavior.
I may take a stab at that in Unreal. 
I *think* the only place in the code that would need to be changed would be:
https://github.com/microsoft/AirSim/blob/a8a2515f1e4c9b54c7a103760cc1413c5d9a562f/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L67
Could search the world actor list for our OriginGeopoint actor and if present would call origin_geopoint.initilize() again on the actor's location. But then would have to offset based on the player-start location. hmm...
Plus I'm not sure what other code has run by the time it gets to the PawnSimApi::setStartPosition method that would rely on the origin_geopoint.

MSBGit, 2020-04-16:
@courageon Thanks! I was thinking you place the _OriginGeopint_ Actor where you want it (either by hand or using the [Transform widget to specify location in ue units](https://docs.unrealengine.com/en-US/Engine/Actors/Transform/index.html)). If the _OriginGeopoint_ Actor exists in the level, use the GPS coordinates specified in the `OriginGeopoint` property in `settings.json` and feed those values to the Pixhawk, otherwise if the  _OriginGeopoint_ Actor does not exist in the level, do what the code does today.

We're both thinking of it the same way.

I don' t know Unreal Editor APIs so can you add a widget or a section to the exiting _Transfer_ widget only for the _OriginGeopoint_ Actor to be able to specify lat/lon/alt, perhaps as a drop down in the _Location_ section which currently has _Relative_ and _Absolute_ so add _GPS_?

> Could search the world actor list for our OriginGeopoint actor and if present would call origin_geopoint.initilize() again on the actor's location. But then would have to offset based on the player-start location. hmm...

So this would un-do the offset to the GPS coordinates that the PlayerStart offset code created?

[image]

courageon, 2020-04-16:
I used to do a fair amount of work in UE3, so I'm hoping at least the same things that were available in UE3 are available in UE4 as well (and from what I've read so far that's most definitely the case). Adding a widget is pretty straight forward, it'll come with all the transforms and basic attributes. You can then add on the extra attributes that you need. And IIRC I think you can tag those attributes with which category they belong under (like "Transform")

> So this would un-do the offset to the GPS coordinates that the PlayerStart offset code created?

It won't undo the offset, the offset will just now be based from the OriginGeopoint actor instead of 0,0,0. Which should mean everything will just line up correctly. *should*

MiraYaziji, 2020-04-22:
Hello, If I have a point in unreal engine whose coordinates are (1000,2000,3000)
If I want to use Airsim APIs to reach this point. The coordinates in the Airsim API should be: (10,20,-30) to reach this point?
And another question, if I want to add an object to unreal engine, I place it then I click on build or build lighting or what else?

Thank you for your help.
Any suggestions @courageon @MSBGit

obitodaitu, 2020-04-22:
@MiraYaziji,it not reach this point.you should read pawnsimapi.cpp,it will convert ue4  coordinate to airsim coordinate.or you can compulate offset related to self.

MiraYaziji, 2020-04-22:
Hello,
 The point is that I didn't understand how to do the offset.
If I have a point having coordinates (a,b,c) in unreal engine, And I want the drone to go to this point. I/m controlling the drone using APIs, so what coordinates should I put when calling the APIs to reach the same point.
Thank you for your help.
@obitodaitu @courageon @MSBGit

MiraYaziji, 2020-04-22:
Please note that I am using python APIs

obitodaitu, 2020-04-22:
@MiraYaziji ,
ASimHUD* CurrentHud = Cast<ASimHUD>(DronePlayerController->GetHUD());
	Vector3r nedPosition = CurrentHud->simmode_->getVehicleSimApi(vehicle)->getNedTransform().toGlobalNed(Position);

MSBGit, 2020-04-24:
@courageon want to work together on the `OriginGeopoint` fix?

MSBGit, 2020-04-24:
Hi MiraYaziji,

> The point is that I didn't understand how to do the offset.
> If I have a point having coordinates (a,b,c) in unreal engine, And I want the drone to go to this point. I/m controlling the drone using APIs, so what coordinates should I put when calling the APIs to reach the same point.

This has been confusing to me as well until I read the following post that describes the relationship between Unreal Units and real-world measurements:

[What is an Unreal Unit?](https://subscription.packtpub.com/book/game_development/9781784394905/1/ch01lvl1sec10/what-is-an-unreal-unit)

This helped me to understand how to use the `simGetObjectPose()` and `simSetObjectPose()` APIs to determine where an object (Actor) is located and then to move that object. Note that the coordinates used by these APIs are relative to the _PlayerStart_ (or _PlayerStartPIE_) Actor. These APIs should also work to control the drone itself, although for my work I am using a Pixhawk 4 in a HITL configuration so I get and set the position of the drone using MAVSDK that talks directly to the Pixhawk.

Your question is what should the value of the coordinates in the AirSim API be to move your drone to a specific location. The APIs use NED SI units (so meters). You need to convert between meters to UE units, and that conversion depends on what value is in your _World to Meters_ setting (by default in Unreal 4.22 and later it's 100 I think).

[AirSim Common APIs](https://github.com/microsoft/AirSim/blob/master/docs/apis.md#common-apis)

courageon, 2020-04-24:
@MSBGit I've got a solution working for the origin geopoint actor. I may toss this over to you if you'd like to look into adding a flag to the settings file to allow switching between setting the geopoint at origin, or setting the geopoint at player-start. Right now I'm going on the premise of "if the OriginGeopoint actor exists in the level, then this level has been pre-designed for a specific location, use that location for geo-positioning in the level". In other words, it's hard-coded at that point and cannot be changed via settings file. But I could see the need to change it via settings file if you just wanted a geo marker pointed at a different location. Maybe add something in the settings file like "IgnoreOriginGeopointActor" or something?

I had a heck of a time trying to track down where in the code to add an offset for the OriginGeopoint actor's location. What I wound up doing is re-basing the geo-location of the origin (0,0,0) based on the location of the OriginGeopoint actor. The same technique should work for re-basing based on the player-start location as well.  

Another thing that completely threw me off for a bit was the NED directions. In the UE's Level Composition this puts North facing left, and East facing up! But you can't tell once you start playing :) 

Oh, and how do you do collaboration on a branch in github? I've never done it before. Should I fork and push what I have then you fork that? Merge, then PR for master?

MSBGit, 2020-04-24:
@courageon wow, that's great! I'd love to test.

Here's the page that describes how to contribute to AirSim:

[Contributing to AirSim](https://github.com/microsoft/AirSim/blob/master/docs/CONTRIBUTING.md)

The project wants us to follow the C++ coding guidelines (style guide) and to file an issue first, which you did [here](https://github.com/microsoft/AirSim/issues/2588).

It looks like the workflow is to fork AirSim into your own github account rather than just creating your own branch in the AirSim repo, so I would follow those instructions:

[The beginner's guide to contributing to a GitHub project](https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/) (note that this isn't the only way to contribute to a github project, it's the way that AirSim and other open source projects do their development workflow).

Then I think I can clone your forked repo and create my own branch locally so I can build and test your changes myself.

----

> I may toss this over to you if you'd like to look into adding a flag to the settings file to allow switching between setting the geopoint at origin, or setting the geopoint at player-start. Right [...]

I won't realistically be able to work on the source code for at least a week (but I can build and test immediately). I'm working on some pre-release MAVSDK issues that need to be resolved soon. I think your premise of "if the OriginGeopoint actor exists int he environment, then use that location for the initial GPS coordinates that are used". That way, existing users don't need to modify their settings file, they just have to place the OriginGeopoint actor where they want the GPS origin to be.

> But I could see the need to change it via settings file if you just wanted a geo marker pointed at a different location. Maybe add something in the settings file like "IgnoreOriginGeopointActor" or something?

Could you handle that case by placing OriginGeopoint where you want it, and that wouldn't affect PlayerStart or the drone stating location? Or am I missing what you're trying to do here?

> I had a heck of a time trying to track down where in the code to add an offset for the OriginGeopoint actor's location. What I wound up doing is re-basing the geo-location of the origin (0,0,0) [...]

Does you mean you moved the origin of the map to OriginGeopoint, so that if I didn't have OriginGeopoint in my environment, the map would be based at some non-zero values, but if OriginGeopoint exists, the map would be based at 0,0,0?

> Another thing that completely threw me off for a bit was the NED directions. In the UE's Level Composition this puts North facing left, and East facing up! But you can't tell once you start playing :)

Interesting. I am setting and getting NED position of my virtual drone using MAVSDK (the getting NED API is what's I'm working on testing since it's not available in MAVSDK master). I'll verify what NED in MAVSDK/PX4 space means compared to what it means in Unreal.

MSBGit, 2020-04-24:
> The point is that I didn't understand how to do the offset.
> If I have a point having coordinates (a,b,c) in unreal engine, And I want the drone to go to this point. I/m controlling the drone using APIs, so what coordinates should I put when calling the APIs to reach the same point.

Try this, this assumes _World to Meters_ is 100:

```
def unrealToAirSimNED(xyz):
    """ Unreal by default is in centimeters and z is positive-up.
    AirSim is in meters and z is positive-down."""
    xyz = xyz * 0.01
    xyz[2] = -xyz[2]
    return xyz
```

MiraYaziji, 2020-04-25:
Okay, Thank you @MSBGit @obitodaitu. 
I will take your propositions into consideration and try to use them

MiraYaziji, 2020-04-25:
To add the object in unreal engine, I just adjust its coordinates? And then what should I do, I click on build or build lighting or what else?
Thank you,
@MSBGit 
@obitodaitu 
@courageon

MSBGit, 2020-04-26:
@MiraYaziji,
> To add the object in unreal engine, I just adjust its coordinates? And then what should I do, I click on build or build lighting or what else?

This kind of question would be better asked in the [Unreal forums](https://forums.unrealengine.com) since it's not really an AirSim issue.

MSBGit, 2020-05-12:
@courageon how's things going with the Origin Geopoint changes?

jonyMarino, 2020-11-11:
Hi @courageon! Thank you for your involvement in the project! We want to fix this issue. If you want to share your code (preferably in a PR), it would be great!

PR #3147 (2020-11-19): Fixed OriginGeopoint GPS error
Fixes #2588, #2575
Currently, "OriginGeopoint" in settings.json is the Lat/Lon location of the world coordinate system origin of the Unreal level  (0,0,0) while it is described in the documentation as the Lat/Lon location of the PlayerStart/Vehicle.

This results in an offset in Lat/Lon between the OriginGeopoint specified and that of the vehicle (reported by the API or by QGC in HITL or SITL):
[image]
With PlayerStart at 
[image], we get:
[image]

This PR fixes the issue by using the PlayerStart location to set a new origin for the world coordinate system. Doing [... cut]
closed 2020-11-19 by jonyMarino