microsoft/AirSim #2590: Combination API flight with remote control flight

rovdrcru, 2020-04-16:
Hello!

I was wondering if it is possible to combine API flight with remote control flight.
To simulate a project as realistic as possible I need to be able to interrupt the API flight program by using my RC (xbox controller). I can't seem to find any documantation about this case. 
Whenever I start a flight program using API's, the remote controller is beeing ignored untill the program has been completed.

How can I make my RC prior to the API program?

I'm using:
- Unreal v24.
- Python 3.8
- Windows

rajat2004, 2020-04-16:
Hmm, sounds interesting, and should be doable, but not too sure
Have you tried using  `enableApiControl(false)` and checking if RC works after that?

jonyMarino, 2020-08-10:
@rovdrcru Have you tried Rajat approach?

closed 2022-03-31 by jonyMarino