microsoft/AirSim #2591: ERROR: cannot launch node of type

EdwinMeriaux, 2020-04-16:
Hello again,

I have another quick issue. On the running step:

roslaunch airsim_ros_pkgs airsim_node.launch;

I get:
ERROR: cannot launch node of type [airsim_ros_pkgs/airsim_node]: Cannot locate node of type [airsim_node] in package [airsim_ros_pkgs]. Make sure file exists in package path and permission is set to executable (chmod +x)

The full output is:

... logging to /home/edwin-pcp/.ros/log/63f83350-7ff0-11ea-affc-1c1b0d61d3f2/roslaunch-edwinpcp-H170N-WIFI-10069.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://edwinpcp-H170N-WIFI:36909/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.001
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [10079]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 63f83350-7ff0-11ea-affc-1c1b0d61d3f2
process[rosout-1]: started with pid [10090]
started core service [/rosout]
ERROR: cannot launch node of type [airsim_ros_pkgs/airsim_node]: Cannot locate node of type [airsim_node] in package [airsim_ros_pkgs]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[ned_to_enu_pub-3]: started with pid [10097]

I apologize. But what did I do wrong?

rajat2004, 2020-04-16:
Did you run `source devel/setup.bash`?

EdwinMeriaux, 2020-04-16:
yes I did.

rajat2004, 2020-04-16:
Maybe try doing a `catkin clean`, build and source again and see what happens
Also check what permissions are there, but don't think a launch file should be executable

EdwinMeriaux, 2020-04-16:
Sorry for the confusion from where should I do catkin clean?

rajat2004, 2020-04-16:
Same directory, all catkin commands have to be run from the `ros` directory

EdwinMeriaux, 2020-04-22:
This solved the issue. Thanks!

closed 2020-04-22 by EdwinMeriaux

NintendoLink, 2021-01-26:
> This solved the issue. Thanks!

i meet the same problem when i run "roslaunch airsim_ros_pkgs airsim_node.launch" ,would you please to share me how you solve it ? thanks

umor3, 2021-01-26:
@NintendoLink did you check if the file is in the folder "./catkin_ws/devel/lib/<package-name>"
Is your CMakeLists.txt-file correct? Does it include `add_executable(<executable_file_name> <path/to/file/file.cpp)`
For example: `add_executable(test_file src/test_file.cpp)`

onurcankaraca, 2021-07-14:
> Sorry for the confusion from where should I do catkin clean?

where did you do "catkin clean"?

WinRobotics, 2021-07-25:
> > Sorry for the confusion from where should I do catkin clean?
> 
> where did you do "catkin clean"?

At the ~/catkin_ws$

onurcankaraca, 2021-07-25:
> > > Sorry for the confusion from where should I do catkin clean?
> > 
> > 
> > where did you do "catkin clean"?
> 
> At the ~/catkin_ws$

Thanks for helping me. I solver my problem.

tqltech, 2022-03-29:
catkin_make clean

mgrallos, 2023-02-16:
I have the same problem and have several packages in my workspace does doing the catkin clean mean I have to build every package again?