microsoft/AirSim #2593: Not able to package Airsim with custom environment

plushpluto, 2020-04-16:
Trying to package Airsim with UE4-4.23 in windows. Getting a lot of exception handling errors. 

error: cannot use 'throw' with exceptions disabled

Please help.

rajat2004, 2020-04-17:
You could try settings `bForceEnableExceptions = true` in [AirSim.Build.cs](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirSim.Build.cs)
Though it already has [`bEnableExceptions = true`](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirSim.Build.cs#L79), maybe they are different? Or the latter one was changed to the first one?

plushpluto, 2020-04-17:
It says
The name 'bForceEnableExceptions' does not exist in the current context

bEnableExceptions = true is already present.

closed 2020-04-19 by plushpluto

abhishekvermaairbus, 2022-07-09:
Any solution to this?