microsoft/AirSim #2598: question :  Korean translation

kimtaeyoung-98, 2020-04-19:
I want to translate these project documents into Korean. Can I contribute to this project with Korean translation?

theold190, 2020-04-22:
@kimtaeyoung-98, it seems that right now the Contributing document only describes code contributions and not translations (see https://github.com/microsoft/AirSim/blob/master/docs/CONTRIBUTING.md).

I haven't found any information about translations to other languages. I have mainly checked the docs section (https://github.com/microsoft/AirSim/tree/master/docs).

Also there is a pull request with an attempt to do a translation that was rejected (https://github.com/microsoft/AirSim/pull/1463). And no details provided on how to do the translation.

A fast Google search of "AirSim translation" didn't show any results.

So, my guess is that project doesn't support the multilingual documentation right now.

jonyMarino, 2020-08-13:
We do not support multi-language documentation.

closed 2020-08-13 by jonyMarino