microsoft/AirSim #2600: Webgl project using Airsim

plushpluto, 2020-04-19:
How to create a webgl project using Airsim?

UE4 or Unity?

rajat2004, 2020-04-20:
AirSim doesn't support WebGL, but you can try to build it yourself. Both Unreal & Unity have documentation regarding this
https://docs.unrealengine.com/en-US/Platforms/HTML5/GettingStarted/index.html
https://docs.unity3d.com/Manual/webgl-building.html

Not all AirSim features are available on Unity however

plushpluto, 2020-04-21:
Webgl is key to my work. I am trying it. Will share how it goes. Need to play around a bit with both Unreal and Unity. Any idea why the support for Unity is less!!

rajat2004, 2020-04-21:
Just needs a contributor to add support for the new APIs, etc

plushpluto, 2020-04-28:
Tried to package Airsim Webgl for the UnityDemo.

Unity-2019.1.0f2 [Ubuntu 18.04/Kernel-5.3.0-46]

Got this error.
To marshal a managed method, please add an attribute named 'monopinvokecallback' to the method definition. the method we're attempting to marshal is: airsim unity.vehicle companion::set pose

Stuck here!!!

rajat2004, 2020-04-28:
I don't have any experience with Unity, Unity forums and Stackoverflow would the best ways to figure out problems
Note that you might need to use the fixes in #2328, to test the branch locally-
```
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git fetch origin pull/2328/head:PR/2328; git checkout PR/2328;
./build.sh; # or build.cmd for Windows
cd Unity
./build.sh; # or build.cmd for Windows
```