microsoft/AirSim #2601: Adding an object to an environment

MiraYaziji, 2020-04-19:
Hello, I cloned this repository: https://github.com/microsoft/AirSim.git
When I press the play button, I obtain the drone in the environment. If I want to add an object to the environment (for example a cube or a sphere) in a specific location and when I run the program, I want that this object appears. If I have the coordinates of this object in Airsim APIs, the location in Unreal Engine should be multiplied by 100? And when I add this object to Airsim what steps should be done?
build geometry or build lighting or what else?
Thank you for your help.

rajat2004, 2020-04-19:
Adding new objects to the environment is all Unreal Engine, and doesn't affect Airsim. There should be plenty of videos, etc. available
As to the coordinate system, see https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#coordinate-system

MiraYaziji, 2020-04-19:
Okay, thank you. I will check how we add objects in Unreal Engine. Concerning this website:
https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#coordinate-system
I checked it, I understood that if in unreal Engine the coordinates of an object are for example (1000,2000,3000) therefore if I want to use AirSim API to reach that object, the coordinates in the AirSim API call should be (10,20,-30)?
Thank you for your help

jonyMarino, 2020-08-19:
Yes, that is correct

closed 2020-08-19 by jonyMarino