microsoft/AirSim #2602: Run active SLAM system on Airsim

haozhu47, 2020-04-19:
I'd like to run active ORB_SLAM2 system on Airsim, but I'm confused  how  Simple PID Position Controller Node works. I 've tried to launch this node but terminal showed wait for odometry even if I have launch airsim_node which provides it. 
It seems services provided by Simple PID Position Controller Node are able to control drones. I cannot understand the docs well, and I doubt whether I need to configure Mavros and PX4. 

Another question about Unreal4 scenes. I download block enviroment from [website](https://github.com/microsoft/AirSim/releases/tag/v1.2.0Linux) , but when exec the ORB_SLAM2, error occurs . When executes the enviroment on UE4, no error shows.

`ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded in 6.79s
Create new map.

Camera Parameters: 
- fx: 336
- fy: 336
- cx: 336
- cy: 188
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 4.99025
OMPL version: 1.4.2
[ERROR] [1587315470.619680566]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
[ERROR] [1587315470.746396180]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
[ERROR] [1587315470.880795029]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
[ERROR] [1587315471.003468894]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
[ERROR] [1587315471.110782676]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
[ERROR] [1587315471.220163417]: cv_bridge exception: Image is wrongly formed: height * step != size  or  376 * 2016 != 1010688
`

rajat2004, 2020-04-19:
v1.2.0 is a very old release, try the latest one

haozhu47, 2020-04-20:
> v1.2.0 is a very old release, try the latest one

Thanks a lot! The second problem is sovled.

jonyMarino, 2021-07-23:
#2602 fixes this.
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

PR #3880 (2021-07-30): fix ROS topic names

Fixes: #3805    
Fixes: #2602
Fixes: #3878 

## About

This PR updates vel_cmd_world_frame and odom_local_ned topic names to include vehicle name so that communication between PIDPositionController and AirsimROSWrapper. The changes for updating the odom_local_ned topic comes from @leoborgnino's work on #2789

## How Has This Been Tested?

ROS node was built with changes applied and the repro steps in #3805 were followed

## Screenshots (if appropriate):
closed 2021-07-23 by jonyMarino