microsoft/AirSim #2606: Use own mathematical representation of vehicle

xtream111, 2020-04-20:
Hello everyone, I am a student trying to conduct a school project regarding autonomous drones.

The objective is to create a model in Simulink and use any simulation tool to simulate this model.
As the objective is to do it autonomously, it is important to have some vision based solution. That is the reason I was looking into Airsim as it can provide a certain degree of photo-realistic environments. This means that Airsim will only provide its "eyes".
Is there any way to do this using, for instace, ROS. Or some built-in functionalities?

Thanks in advance

rajat2004, 2020-04-20:
Not sure as to what you're looking for, but ROS wrapper is there - https://github.com/microsoft/AirSim/blob/master/docs/airsim_ros_pkgs.md
Other options are the Python and C++ APIs which you would already know about

xtream111, 2020-04-21:
Thank you for your response, I don't know if I explained myself correctly.
If you check the architecture of Airsim, you get this 
[image]

What I want to do is to change the vehicle model, so I can plug my own model form Simulink.

rajat2004, 2020-04-22:
Don't think that'll work, the vehicle model will need to be modified in UE itself rather than plugging in an external one
But the maintainers would have a better idea about this

jonyMarino, 2021-04-30:
You can accomplish this using the new ExternalPhysicsEngine: #3626

Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-04-30 by jonyMarino