microsoft/AirSim #2607: multithreading in airsim using the same drone

EdwinMeriaux, 2020-04-20:
Hi,

So when I send commands to the drone I want to know it's position as it is moving between two points. I could not find a way not using multithreading to do this (if there is please let me know).

So using 1 drone the first thread just gives it commands and measures the points when completing a movement between two points and the other thread measures the points between the movements (for the time being it will only measure one point cause even that is not working)

here is the code:
import setup_path 
import airsim

import numpy as np
import os
import tempfile
import pprint
import cv2
import math
import time
import threading

class Position:
    def __init__(self, pos):
        self.x = pos.x_val
        self.y = pos.y_val
        self.z = pos.z_val

class Position2:
    def __init__(self, pos2):
        self.x = pos2.x_val
        self.y = pos2.y_val
        self.z = pos2.z_val

pxtol = []
pytol = []
pztol = []

px2tol = []
py2tol = []
pz2tol = []

pivals = np.arange(0,2.01,0.01)
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)	
client.armDisarm(True)
client.takeoffAsync().join()
client.moveToPositionAsync(-10, 10, -10, 5).join()

def tracking():
	var = 0
	pos2 = client.getMultirotorState().kinematics_estimated.position
	px2tol.append(pos2.x_val)
	py2tol.append(pos2.y_val)
	pz2tol.append(pos2.z_val)
	print ("pos2x",pos2.x_val)
	print ("pos2y",pos2.y_val)
	print ("pos2z",pos2.z_val)
	#print(var)

def moving():
	Position = client.getMultirotorState().kinematics_estimated.position
	print(Position)
	pos = client.getMultirotorState().kinematics_estimated.position
	print(pos.z_val)
	client.moveToPositionAsync(0, 0, -20, 5).join()
	Position = client.getMultirotorState().kinematics_estimated.position
	print(Position)
	
	for i in pivals:
		client.moveToPositionAsync(10*math.sin(math.pi*i),10*math.cos(math.pi*i),-20, 5).join()
		pos = client.getMultirotorState().kinematics_estimated.position
		pxtol.append(pos.x_val)
		pytol.append(pos.y_val)
		pztol.append(pos.z_val)	
		print(pos)
		time.sleep(2)
	print("moved")
	client.hoverAsync().join()
	print("hover")
	Position = client.getMultirotorState().kinematics_estimated.position
	print(Position)
	client.enableApiControl(False)

t1 = threading.Thread(target=moving) 
t2 = threading.Thread(target=tracking)

t1.start()
t2.start() 

print("pxtol")
print(pxtol)
print("pytol")
print(pytol)
print("pztol")
print(pztol)

but I keep getting this error:

Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

pxtol
[]
pytol
[]
pztol
[]
<Vector3r> {   'x_val': -10.107412338256836,
    'y_val': 10.109457015991211,
    'z_val': -10.045998573303223}
-10.045998573303223
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "Edwin3.py", line 45, in tracking
    pos2 = client.getMultirotorState().kinematics_estimated.position
  File "/home/edwin-pcp/AirSim/PythonClient/airsim/client.py", line 658, in getMultirotorState
    return MultirotorState.from_msgpack(self.client.call('getMultirotorState', vehicle_name))
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 25, in get
    self.join()
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 22, in join
    self._loop.start()
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/loop.py", line 22, in start
    self._ioloop.start()
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/tornado/ioloop.py", line 755, in start
    raise RuntimeError("IOLoop is already running")
RuntimeError: IOLoop is already running

<Vector3r> {   'x_val': 0.08782163262367249,
    'y_val': -0.13658884167671204,
    'z_val': -19.997657775878906}
<Vector3r> {   'x_val': 0.04991109296679497,
    'y_val': 10.124534606933594,
    'z_val': -19.897062301635742}
<Vector3r> {   'x_val': -0.4412935972213745,
    'y_val': 11.106693267822266,
    'z_val': -19.902856826782227}
<Vector3r> {   'x_val': 0.20392726361751556,
    'y_val': 9.808135986328125,
    'z_val': -19.930095672607422}
<Vector3r> {   'x_val': -0.0948425754904747,
    'y_val': 10.45771598815918,
    'z_val': -19.916133880615234}
<Vector3r> {   'x_val': -0.00489964010193944,
    'y_val': 10.204997062683105,
    'z_val': -19.910804748535156}
<Vector3r> {   'x_val': -0.016243308782577515,
    'y_val': 10.28304672241211,
    'z_val': -19.909713745117188}
and then it keeps on giving the data points from only the first thread

I tried only running the second thread and in that case I get this:
WARNING:tornado.general:Connect error on fd 8: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 8: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 8: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 8: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 8: ECONNREFUSED
Traceback (most recent call last):
  File "Edwin3.py", line 36, in <module>
    client.confirmConnection()
  File "/home/edwin-pcp/AirSim/PythonClient/airsim/client.py", line 61, in confirmConnection
    if self.ping():
  File "/home/edwin-pcp/AirSim/PythonClient/airsim/client.py", line 24, in ping
    return self.client.call('ping')
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/edwin-pcp/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit

So can I not measure the data point of the same drone in two different threads at the same time?

rajat2004, 2020-04-21:
Have you tried using separate clients for each thread?

EdwinMeriaux, 2020-04-21:
I did not know I could do that. 
How would I do that?

rajat2004, 2020-04-21:
Think just passing the clients as args to the functions might work
```
cl1 = airsim.VehicleClient()
cl2 = airsim.VehicleClient()

threading...args=(cl1))
threading...args=(cl2))
```

EdwinMeriaux, 2020-04-22:
Thanks! That solved the issue!

closed 2020-04-22 by EdwinMeriaux

ME-Msc, 2024-06-19:
> Hi, if I want use Airsim's client in the main and image_process threads at the same time, it will be a problem cause the client is occupied, how should I solve this problem to speed up my program. Thank you!

Hi, have you solved it? I met the same problem, thanks for your help! 

I run a thread to control the uav fly, and the other thread to get image from the camera. The client is occupied.

ME-Msc, 2024-06-19:
@paobuliao How fast you reply! Thanks!

Luckily, I fixed it a few minutes ago, and I made a simple test. It works! The two thread could run individually, take photo while flying to a position. Here is the code. Hope it is useful for you.

	import airsim
	import threading

	global_camera = "GlobalCamera"
	client.enableApiControl(True, global_camera)
	client.takeoffAsync(vehicle_name=global_camera).join()
	client.moveToPositionAsync(0, 0, -40, 10, vehicle_name=global_camera).join()
	airsim.wait_key()
	
	def fly():
		client1 = airsim.MultirotorClient()
		res = client1.moveToPositionAsync(20,20,20,2,vehicle_name="GlobalCamera")
		res.join()

	def photo():
		client2 = airsim.MultirotorClient()
		response = client2.simGetImage("bottom_center", airsim.ImageType.Scene, vehicle_name="GlobalCamera")	

		import os
		current_file_path = os.path.abspath(__file__)
		current_directory = os.path.dirname(current_file_path)
		import datetime
		time_str = datetime.datetime.now().strftime("%Y%m%d%H%M%S")
		vehicle_directory = os.path.join(current_directory, "data", "GlobalCamera")
		filename = os.path.join(vehicle_directory, f"{time_str}.png")
		if not os.path.exists(vehicle_directory):
			os.makedirs(vehicle_directory)
		with open(filename, "wb") as f:
			f.write(response)

	threads = []
	threads.append(threading.Thread(target=fly, ))
	threads.append(threading.Thread(target=photo, ))

	# start all threads
	for thread in threads:
		thread.start()

	# wait for all threads finish
	for thread in threads:
		thread.join()