microsoft/AirSim #2608: Unreal Editor

nightduck, 2020-04-20:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

rajat2004, 2020-04-21:
Hmm, sounds strange, does this happen without running the ROS package, ie. just the Editor? Also, which version of UE are you using, 4.24?
You could also try using OpenGL instead of the Vulkan drivers which UE now uses by default. Vulkan does take up more memory right now, but hopefully, some of the memory issues might be fixed in UE 4.25
For using OpenGL, just use the `-opengl4` option while launching Editor from the terminal

nightduck, 2020-04-21:
If I run the blocks simulation totally as is, it'll run fine. When I copy over the `front_stereo_and_center_mono.json` into `Documents/Airsim/settings.json` and the lidar dots show up, that's when the problem appears. At first, it appeared to be a low frame-rate issue, but the fix discussed in issue #1185 did nothing.

Yes, UE 4.24

I wasn't aware openGL was an option. Before I had vulkan installed, UE complained it couldn't use openGL, but I'll try that flag.

rajat2004, 2020-04-21:
Yes, sorry forgot to mention, follow the first comment in https://www.reddit.com/r/unrealengine/comments/e4i7ny/cannot_start_unreal_engine_4_in_linux/

nightduck, 2020-04-21:
Unclear on how to delete cameras

rajat2004, 2020-04-21:
Just remove from the settings if not required

nightduck, 2020-04-21:
The fix was to go into settings.json and set `DrawDebugPoints` to false. The debug points cumulatively add up and bog down Unreal Engine.

Consider this issue now a feature request: Lidar, by default, doesn't not render all points ever captured, only recent ones, say from the last frame.

rajat2004, 2020-04-21:
This problem didn't occur with me when I was using Lidar. But it could be the case that I just didn't leave it on that much
Now that I think about it, I haven't actually used it on UE 4.24, so it might be different. How much time did it take to hog up the memory and crash the system?

nightduck, 2020-04-22:
It would take about 5-10 seconds for the framerate to drop to rock bottom. The render time for each frame increases linearly, so the fps has a 1/x relationship. Once it gets down to 2-3fps, it gobbles up a gig of ram every 3-4 seconds or so. If your system doesn't have swap space, only UE will crash, not the OS. Either way, it's directly related to memory usage.

And [both](https://github.com/microsoft/AirSim/blob/266cd4333a8a1aa3abc5f189917c53808392b123/ros/src/airsim_tutorial_pkgs/settings/front_stereo_and_center_mono.json#L31) the [settings](https://github.com/microsoft/AirSim/blob/266cd4333a8a1aa3abc5f189917c53808392b123/ros/src/airsim_tutorial_pkgs/settings/two_drones_camera_lidar_imu.json#L25) files that specify DrawDebugPoints default it to true.

Which version of UE are you using? I can build that and test it on my system. I know that [UE switched to Vulkan in 4.21](https://www.phoronix.com/scan.php?page=news_item&px=Unreal-Engine-4.21-Debuts). Anything earlier would use OpenGL, which doesn't have the memory bug, only low frame rates.

rajat2004, 2020-05-01:
#2614 has been merged, so I think this issue can be closed

nightduck, 2020-05-01:
Agreed

closed 2020-05-01 by nightduck