microsoft/AirSim #2609: Unity: Editor quits after selecting a vehicle "Problem in starting AirSim server!!!"

conorokane, 2020-04-21:
To reproduce: Follow the install instructions here: https://microsoft.github.io/AirSim/Unity/
Play the SimModeSelector scene.
Click the Car Mode or Drone Mode button.
Error message appears: 
`Problem in starting AirSim server!!! Please check logs for more information`
The editor then quits.

Editor log shows: 
```
UnityEngine.UnhandledExceptionHandler:<RegisterUECatcher>m__0(Object, UnhandledExceptionEventArgs) (at C:\buildslave\unity\build\Runtime\Export\Scripting\UnhandledExceptionHandler.bindings.cs:46)
 
(Filename: Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs Line: 139)
```

Tested in Unity 2018.2.15f1 and 2019.1.0f2 on Windows 10.

rajat2004, 2020-04-21:
This is probably due to a current bug in the code, which is being fixed in https://github.com/microsoft/AirSim/pull/2328

TerryTeng88, 2020-04-22:
I have the same error: Problem in starting AirSim server!!! Please check logs for more information 
unity2019.3.8f1 windows10

cm1405276135, 2020-04-23:
Hello,Have you solved the problem?

rajat2004, 2020-04-23:
If someone would like to try the PR, then use the following commands
```
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git clean -fdx
git fetch origin pull/2328/head:PR/2328; git checkout PR/2328;
cd Unity
./build.sh; # or build.cmd for Windows
```

conorokane, 2020-04-23:
I tried pulling #2328 and building but the build failed:

```
"D:\Documents\Projects\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (Clean;Build target) (1) ->
"D:\Documents\Projects\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (default target) (2:2)
->
"D:\Documents\Projects\AirSim\AirLib\AirLib.vcxproj" (default target) (3:4) ->
(ClCompile target) ->
  D:\Documents\Projects\AirSim\AirLib\include\common\VectorMath.hpp(12,10): fatal error C1083: Cannot open include file
: 'Eigen/Dense': No such file or directory [D:\Documents\Projects\AirSim\AirLib\AirLib.vcxproj]
```

rajat2004, 2020-04-23:
Did you run `setup.cmd` once from the root directory?

conorokane, 2020-04-23:
> Did you run build.cmd once from the root directory?

No I forgot about that! **It works now, tested in Unity 2019.2.11f1.**

For anyone else wanting to try branch #2328 here are the steps:
```
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git fetch origin pull/2328/head:PR/2328; git checkout PR/2328;
./build.sh; # or build.cmd for Windows
cd Unity
./build.sh; # or build.cmd for Windows
```

rajat2004, 2020-04-23:
Glad to know it works. 
The `build.cmd` from the root directory is only needed the first time, since it downloads some external libraries such as Eigen. After that, just running `build.cmd` from Unity is needed after any code changes

fzambetta-rmit, 2020-04-24:
@rajat2004 , I am assuming this branch delivers just the bug fixing highlighted by @conorokane .
Is the Unity API coverage information still accurately reflected [here ](https://microsoft.github.io/AirSim/unity_api_support/) (i.e., Unity is not at feature parity with UE)?

rajat2004, 2020-04-24:
I think the API list is correct, there might be some more APIs not present in the list which haven't been implemented yet in Unity

TerryTeng88, 2020-04-24:
The unity version I ended up using was 2018.2.15f1. Now it's up and running。

rajat2004, 2020-05-01:
#2328 has been merged, so I think this issue can be closed

madratman, 2020-05-01:
The "Problem in starting airsim server" happens if settings.json has a non-empty "Vehicles" section. 

So this fails: 
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
   "Vehicles": {
      "Drone1": {
          "VehicleType": "simpleflight"
          }
      }
  }
```

But this works perfectly fine:
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}
``` 

This error is still there post #2328

rajat2004, 2020-05-01:
This is probably a problem with the way Unity handles the settings, there's a PR https://github.com/microsoft/AirSim/pull/2303 which fixes problems with Unity, but I haven't tested it yet and it's quite an extensive PR to review as well

TomSeestern, 2020-05-04:
Workaround for this Issue:
[from jasonhbartlett](https://github.com/microsoft/AirSim/issues/1617#issuecomment-447483295)

I think this Workaround is currently needed to get SITL working with PX4 <-> AirSim <-> Unity