microsoft/AirSim #2616: Airsim/ROS tutorial issue

EdwinMeriaux, 2020-04-23:
I am running through the tutorial to get Airsim with ROS. Here is what I am following:
https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/

In the first example:
Single drone with monocular and depth cameras, and lidar

Everytime I run:
$ roslaunch airsim_ros_pkgs airsim_node.launch;

I am able to connect:

... logging to /home/edwin-pcp/.ros/log/3c9086fc-858e-11ea-b012-1c1b0d61d3f2/roslaunch-edwinpcp-H170N-WIFI-4875.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://edwinpcp-H170N-WIFI:36683/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.001
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [4885]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3c9086fc-858e-11ea-b012-1c1b0d61d3f2
process[rosout-1]: started with pid [4896]
started core service [/rosout]
process[airsim_node-2]: started with pid [4903]
process[ned_to_enu_pub-3]: started with pid [4904]
simmode_name: Multirotor
Imu
Lidar
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
AirsimROSWrapper Initialized!

but then when I need to launch:
roslaunch airsim_tutorial_pkgs front_stereo_and_center_mono.launch 

I get:
RLException: [front_stereo_and_center_mono.launch] is neither a launch file in package [airsim_tutorial_pkgs] nor is [airsim_tutorial_pkgs] a launch file name
The traceback for the exception was written to the log file

Is something not set up properly?

rajat2004, 2020-04-24:
Did you run `source devel/setup.bash`?

EdwinMeriaux, 2020-04-24:
Yes I did. Just to confirm you mean running that (source devel/setup.bash) is PATH_TO/airsim/ros/devel right?

rajat2004, 2020-04-24:
I think you forgot to run that when a new terminal was opened. This has to run in every terminal you're running the ROS commands from, rather than just the first terminal since the changes are not carried across to the new ones

EdwinMeriaux, 2020-04-24:
I misunderstood I thought you meant: did I run that in the initial terminal. Yes I forgot to do that in the second terminal and yeah that solved the issue!

Thank you so much!

closed 2020-04-24 by EdwinMeriaux