microsoft/AirSim #2617: Variable shadowing error when packaging project

ndalton12, 2020-04-23:
So I am interested in building a binary for my team on linux of a customized version of the blocks environment. However, when I try packaging the build I get an error about variable shadowing. 

Apparently this was raised form warning to error in UE 4.24 as seen [here](https://forums.unrealengine.com/development-discussion/c-gameplay-programming/1699015-4-24-c-transition-guide). I tried changing the Blocks.target.cs and BlocksEditor.target.cs under the project source to set the shadow variable level back to warning, but nothing changes. Any advice? I couldn't find anything on this on the unreal forums so asking here.

> [2020.04.23-20.33.28:697][ 36]UATHelper: Packaging (Linux):   In file included from /home/niall/code/AirSim/Unreal/Environments/Blocks 4.24/Intermediate/Build/Linux/B4D820EA/Blocks/Development/GeometryCollectionEngine/Module.GeometryCollectionEngine.2_of_2.cpp:5:
[2020.04.23-20.33.28:697][ 36]UATHelper: Packaging (Linux):   In file included from /home/niall/code/UnrealEngine/Engine/Source/Runtime/Experimental/GeometryCollectionEngine/Private/GeometryCollection/PhysicsAssetSimulation.cpp:14:
[2020.04.23-20.33.28:697][ 36]UATHelper: Packaging (Linux):   In file included from Runtime/Engine/Classes/Components/SkeletalMeshComponent.h:16:
[2020.04.23-20.33.28:697][ 36]UATHelper: Packaging (Linux):   /home/niall/code/UnrealEngine/Engine/Source/Runtime/Engine/Public/Animation/AnimCurveTypes.h:164:3: error: declaration shadows a variable in namespace 'Chaos' [-Werror,-Wshadow]
[2020.04.23-20.33.28:698][ 36]UATHelper: Packaging (Linux):                   X = 0,
[2020.04.23-20.33.28:698][ 36]UATHelper: Packaging (Linux):                   ^
[2020.04.23-20.33.28:698][ 36]UATHelper: Packaging (Linux):   /home/niall/code/UnrealEngine/Engine/Source/Runtime/Experimental/Chaos/Public/Chaos/ParticleDirtyFlags.h:12:3: note: previous declaration is here
[2020.04.23-20.33.28:698][ 36]UATHelper: Packaging (Linux):                   X =                              1 << 0,
[2020.04.23-20.33.28:698][ 36]UATHelper: Packaging (Linux):                   ^
[2020.04.23-20.33.28:698][ 36]PackagingResults: Error: declaration shadows a variable in namespace 'Chaos' [-Werror,-Wshadow]
[2020.04.23-20.33.35:949][586]UATHelper: Packaging (Linux):   1 error generated.
`

rajat2004, 2020-04-24:
This is a bug in the Unreal Engine code, which has been fixed in the upcoming 4.25 release
https://github.com/EpicGames/UnrealEngine/commit/cb388710a7fbe43eaa82a6d8c43b1632f25f6386
You can manually change the code and package

ndalton12, 2020-04-24:
> This is a bug in the Unreal Engine code, which has been fixed in the upcoming 4.25 release
> [EpicGames/UnrealEngine@cb38871](https://github.com/EpicGames/UnrealEngine/commit/cb388710a7fbe43eaa82a6d8c43b1632f25f6386)
> You can manually change the code and package

Great, thanks! Closing this now.

closed 2020-04-24 by ndalton12