microsoft/AirSim #2622: Error when trying to talk to px4 through AirSim

ndalton12, 2020-04-24:
I'm trying to run px4 controller through AirSim. I have gone through the instructions and my RC is able to transmit to the px4; I can tell because I successfully calibrated the RC in QGroundControl. However, when I run the basic blocks environment with 

> {
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor"
    }
  }
}

I get an error saying it can't connect to the serial port.

`Could not open the serial port /dev/ttyACM0, error=-1`

Any advice? I checked to make sure that /dev/ttyACM0 is correct and that I have correct permissions already.

MSBGit, 2020-04-25:
I run AirSim on Windows. Is the settings.json file you posted your complete file? It seems to be missing parameters related to the serial port you want to use. Here is my settings.json file:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "SerialPortBaudRate": "115200",
      "SerialPort": "COM3",
      "UseSerial": true,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "LocalHostIp": "127.0.0.1"
    }
  },
  "OriginGeopoint": {
    "Latitude": 37.776580,
    "Longitude": -122.309579,
    "Altitude": 0
  }
}
```
Are you sure that your Pixhawk is on `/dev/ttyACM0`? Try having QGroundControl talk to the Pixhawk on that port to verify that it's correct. Also remember that you can't have both QGC and AirSim using the same serial port to talk to the Pixhawk. In my setup, AirSim talks to the Pixhawk over the serial port mapped to the USB interface and I have QGC talk to the Pixhawk either through a telemetry radio link or via a UDP port into AirSim. Both seem to work OK.

Here is a link to a doc that I am working on that shows how to get AirSim and Pixhawk up ad running in HITL mode; it assumes you're running everything on Windows but I don't think there are any significant differences if you are running all this on Linux. It's a WIP so feel free to comment on it: [HITL Setup Pixhawk 4 with AirSim](https://docs.google.com/document/d/19wkUL27XO5ic3ycEKqKD56y6bGnnAwyaBiALWcd0Pzs/edit?usp=sharing).

ndalton12, 2020-04-26:
@MSBGit Thanks for the response. The missing parameters just default to their values as described in [settings page](https://microsoft.github.io/AirSim/settings/). I can connect to the px4 on `/dev/ttyACM0` through QGC and get responses from my radio. I didn't realize that QGC shouldn't be trying to communicate at the same time through the same serial port, so closing it entirely seems to help. I don't have a telemetry radio ATM and the UDP port seems to crash AirSim. Despite that, I did get it to run a couple times after closing QGC and disabling QGC's autoconnect. Not sure if the behavior is deterministic, need to do some more testing. 

BTW, any ideas how to connect to the px4 console without QGC? I can connect to it inside QGC, but running `screen /dev/ttyACM0 115200 8N1` repeatedly fails under different conditions.

Going to close this for now, anyone feel free to reopen if the issue emerges again.

closed 2020-04-26 by ndalton12

ndalton12, 2020-04-26:
Update: Now I am having a different problem, where sometimes I get a GPS lock and sometimes I don't. And sometimes after getting the GPS lock, commands from the radio don't register still. The only fix I have right now is to re-connect to QGC, disconnect from QGC, and re-reun your AirSim. It takes a couple tries usually for this to work, although sometimes it works right away. Very strange, no idea what's going on.

MSBGit, 2020-04-26:
> @MSBGit Thanks for the response. The missing parameters just default to their values as described in [settings page](https://microsoft.github.io/AirSim/settings/).

I have found that what the AirSIm docs say and what the code actually does is not always the same. For example, the [OriginGeopoint issue](https://github.com/microsoft/AirSim/issues/2588) that I am working with @courageon on (docs say one thing, code does something else). So after working with AirSim since last year I take the approach "trust but verify" when it comes to reading the documentation.

> I can connect to the px4 on `/dev/ttyACM0` through QGC and get responses from my radio.

Which radio? You mean your telemetry radio?

> I didn't realize that QGC shouldn't be trying to communicate at the same time through the same serial port, so closing it entirely seems to help.

Yep, I ran into the same thing when I first started working with this. I think there is a MAVLink multiplexer that will allow multiple MAVLink clients to talk to a single Pixhawk but I've never used it.

> I don't have a telemetry radio ATM and the UDP port seems to crash AirSim.

The UDP port method works well with AirSim on Windows, but maybe HITL on Linux has some issues. I'll be spinning up an Ubuntu box to run AirSim and Unreal on to compare to Windows to see what works and what doesn't.

If you don't have a telemetry radio, you can get a USB to serial adapter and wire it into the TELEM1 port on the Pixhawk; make sure your adapter is 3.3V and not 5V or 12V. Then AirSim talks to one serial port and QGC talks to the other. I haven't tried this myself but I have connected a companion computer (Jetson Nano) to the TELEM2 port and I talk MAVLink (using [MAVSDK](https://github.com/mavlink/MAVSDK)) to the Pixhawk.

> Despite that, I did get it to run a couple times after closing QGC and disabling QGC's autoconnect. Not sure if the behavior is deterministic, need to do some more testing.

I always disable QGC's auto connect so that it doesn't surprise me and try to connect to the wrong thing.

> BTW, any ideas how to connect to the px4 console without QGC? I can connect to it inside QGC, but running `screen /dev/ttyACM0 115200 8N1` repeatedly fails under different conditions.

Do you want to talk to the system console of the Pixhawk? This link: [PX4 System Console](https://dev.px4.io/v1.9.0/en/debug/system_console.html) describes how to do that using an FTDI (USB to serial) cable.

If you want to talk to the MAVLink shell (what you would do via QGC), this link: [MAVLink Shell](https://dev.px4.io/v1.9.0/en/debug/system_console.html#mavlink_shell) describes how to do that.

Hope this helps. It would be great to get updates here in how you make out.

ndalton12, 2020-04-27:
Good point about the docs, I'll be sure to double check with the code next time. By radio I really just meant my RC.

You could probably get UDP on linux working, I just tried it once and didn't mess around with it too much. Thanks for the other advice! I got MAVLink shell running but it hangs on the connection for some reason. 

I also discovered what the problem is when the RC commands appear not to be registering at first; turns out you need to have the throttle all the way down before attempting to arm the drone. I arm the drone with one my toggle switches as moving both sticks down and in doesn't seem to work. I wasn't aware of throttle issue until poking around in QGC more.