microsoft/AirSim #2623: build.cmd failed in building AirSim

AngelTang190, 2020-04-25:
I am very new to coding and I am trying to build AirSim in Windows 10 using x64 Native Command Prompt for VS 2019. I cloned the repo for AirSim and I found an error:

fatal: could not create work tree dir 'AirSim'.: Permission denied

I used this and it works
cd \
mkdir code
cd code
git clone https://github.com/Microsoft/AirSim.git

After that, I go to cd\code\AirSim and run build.cmd. Then it gives me the below errors:

Build FAILED.

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(Link target) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\code\AirS
im\DroneShell\DroneShell.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\HelloDrone\HelloDrone.vcxproj" (default target) (5) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\code\AirS
im\HelloDrone\HelloDrone.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\DroneServer\DroneServer.vcxproj" (default target) (6) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\code\AirS
im\DroneServer\DroneServer.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\Examples\Examples.vcxproj" (default target) (8) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\code\AirS
im\Examples\Examples.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\HelloCar\HelloCar.vcxproj" (default target) (12) ->
  LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\code\AirS
im\HelloCar\HelloCar.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(Link target) ->
  AirLib.lib(client.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value
'2' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(client.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't mat
ch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match val
ue '2' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't
match value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(response.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match valu
e '2' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(response.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't m
atch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(format.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value
'2' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  AirLib.lib(format.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't mat
ch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\DroneShell\DroneShell.vcxproj]
  C:\code\AirSim\DroneShell\build\x64\Debug\DroneShell.exe : fatal error LNK1319: 8 mismatches detected [C:\code\AirSim
\DroneShell\DroneShell.vcxproj]

"C:\code\AirSim\AirSim.sln" (default target) (1) ->
"C:\code\AirSim\HelloDrone\HelloDrone.vcxproj" (default target) (5) ->
  AirLib.lib(client.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value
'2' in main.obj [C:\code\AirSim\HelloDrone\HelloDrone.vcxproj]
  AirLib.lib(client.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't mat
ch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\HelloDrone\HelloDrone.vcxproj]
  AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match val
ue '2' in main.obj [C:\code\AirSim\HelloDrone\HelloDrone.vcxproj]
[... 78 lines cut ...]
ue '2' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(rpc_error.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't
match value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(response.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match valu
e '2' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(response.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't m
atch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(format.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value
'2' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  AirLib.lib(format.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't mat
ch value 'MDd_DynamicDebug' in main.obj [C:\code\AirSim\HelloCar\HelloCar.vcxproj]
  C:\code\AirSim\HelloCar\build\x64\Debug\HelloCar.exe : fatal error LNK1319: 8 mismatches detected [C:\code\AirSim\Hel
loCar\HelloCar.vcxproj]

    5 Warning(s)
    53 Error(s)

Time Elapsed 00:00:25.22

#### Build failed - see messages above.

For the warnings, I tried using /NODEFAULTLIB and add in MSVCRT in the linker definition but it still does not work as well.
I tried defining in the preprocessor _allow_iterator_debug_level_mismatch , however it still gives me back error LNK2038.

Can anyone be so kind to tell me what should I do to run build.cmd successfully? Thank you!

rajat2004, 2020-04-25:
I think the problem might be due to the x64 Command Prompt, you need to use the Developer Command Prompt for VS 2019

AngelTang190, 2020-04-25:
> I think the problem might be due to the x64 Command Prompt, you need to use the Developer Command Prompt for VS 2019

I tried using the Developer Command Prompt for VS 2019 but it still gives me the same error message.

rajat2004, 2020-04-25:
Not idea, I'm unable to reproduce this. Just to make sure, you're using the latest master correct?

AngelTang190, 2020-04-25:
> Not idea, I'm unable to reproduce this. Just to make sure, you're using the latest master correct?

I am following the instructions given on this website.
https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md

AngelTang190, 2020-04-26:
I deleted everything starting from the code folder and clone it again and repeat the same process. It worked fine now.

closed 2020-04-26 by AngelTang190

heyBryceZhang, 2024-06-21:
> I deleted everything starting from the code folder and clone it again and repeat the same process. It worked fine now.

I delete all files of Airsim folder, clone the code agian, and run build.cmd, all worked fine. 
I think the previous issue was caused by running build.cmd before UE was fully  installed.