microsoft/AirSim #2626: The following module is missing or built with a different engine version Airsim?

winter2897, 2020-04-26:
OS: Window 10; Airsim: latest version; Python: 3.6; Unreal: 4.24.3

I successfully built.cmd and proceeded to drop the Unreal \ Plugins folder into my project to use the Airsim plugin integration. However, I still encounter the error below despite the successful build.
[image]
[image]

rajat2004, 2020-04-26:
`build.cmd` downloads some external libraries also, and puts everything in the Plugins folder
After copying the Plugins folder, UE will need to build the Engine specific code, which is being shown in the prompt
Are you getting any error while building it?

MSBGit, 2020-04-26:
I see this when I switch between running my environment in the VS debugger (using code built with debugging enabled) and when I run my environment just in the editor (Unreal Editor) by double clicking on the project file.

winter2897, 2020-04-27:
I did not encounter any errors!

Thanks & best regards
-------------------------------------------------------------------------
Tran Hai Quan
School of Electronics and Telecommunications (SET-K60)
Hanoi University of Science and Technology (HUST)
Mobile: 0335247278

winter2897, 2020-04-27:
[image]
[image]

rajat2004, 2020-04-27:
The logs are stored in `Saved/Logs`, which you can see to figure out where the error is and upload them
Else you could right-click on the .uproject file, Generate Visual Studio files, and open the .sln file in VS 2019 and build from there, which will show any warnings or errors while compiling

hrnbot, 2020-04-28:
I had same issue but problem was I had pre-installed visual studio 2017 and visual studio 2019. Check file Saved/logs if you spot anywhere 2017 that means it is build in 2017. 

So, just remove 2017 and build using 2019 only.

Thanks @rajat2004 for giving advice to che k saved/logs

winter2897, 2020-04-30:
[image]
my logs

winter2897, 2020-04-30:
The problem has been resolved. I have not updated the .NET SDK yet.

closed 2020-04-30 by winter2897