microsoft/AirSim #2627: Confusion in the workflow

plushpluto, 2020-04-27:
Sorry guys, had to post it here since I am really confused on the basics. Few questions!!!

1. I am trying to test the workflow of the SITLs available mainly using MAVSDK/Dronekit/MAVROS and understand the robustness of them. Dronekit --> PX4/ArduPilot --> Airsim --> UnrealEngine. 
Is my understanding of the workflow correct? Is there any way to temporarily bypass Airsim?

2. Can I disable any sensor using the Airsim client during the flight? If not, is there any way to do it other than the settings.json file?

rajat2004, 2020-04-27:
The workflow of SITL seems correct. Not sure as to what you mean by bypassing Airsim, any UE specific tasks such as adding new actors, or changing the environment can be done as usual in UE

AFAIK, there's no way to disable sensors at runtime, maybe you could add an API for such a task, though will need to look into how it can be done

plushpluto, 2020-04-27:
Got it. Workflow seems clear to me now. Thanks.
I am working on a webgl based project for Airsim. I need to figure out a way to use Airsim as a separate service not an integrated one with the UE. Let UE be based out on MAVLink. It would be better if the workflow is Airsim --> MAVLink --> UE. Just a thought. I don't know. I may be wrong.

Ah! man. Sad to know that there is no way to disable the sensors!

closed 2020-08-10 by jonyMarino