microsoft/AirSim #2628: Getting pose of moving objects

manthan99, 2020-04-27:
I am using the Africa (V 1.2.0 -Linux) environment. My work requires to know the pose of the moving animals. To get the pose of the animals, I need to specify the name/tag of the actor in the `simGetObjectPose` API. 
Since only the .pak file is available for the Africa environment, I cannot open it in the UE4 editor. Is there any method to get to know the names of all actors present in the environment so that I can use the `simGetObjectPose` API to get the ground truth of the animals?

I am using UE 4.24 on a Ubuntu 18.04 system

rajat2004, 2020-04-27:
v1.2 is a very old release, try to use the latest ones
`simListSceneObjects` can be used to get the names of all the actors

manthan99, 2020-04-27:
Thank you very much. It seems to solve the issue. Any idea when will the Africa environment be available for the current release in Linux?

rajat2004, 2020-04-27:
No idea, the maintainers would know. Compiling and uploading binaries is a pretty compute and time-intensive task, maybe why it's taking a bit long

rajat2004, 2020-04-29:
Africa and many other envs are now available for the Linux release

closed 2020-04-29 by manthan99