microsoft/AirSim #2629: Crashes while start play CarDemo or DroneDemo

hrnbot, 2020-04-27:
I am using unity 2017.2.15f1
visual studio 2019 with windows SDK 10.0.18362.0

- Start x64 Native Tools Command Prompt for VS 2019.

- Clone the repo: git clone https://github.com/Microsoft/AirSim.git, and go the AirSim directory by cd AirSim.

- Run build.cmd from the command line.

- Go inside the AirSim\Unity directory: cd Unity.

- Build the unity project: build.cmd.

- Start Unity and click Open project.

- Select the folder AirSim\Unity\UnityDemo, and then hit the button Select Folder.

- In the bottom pane, Click on Projects->Assets->Scenes. Then, Double-click on SimModeSelector. This will load the SimModeSelector scene into the scene hierarchy pane. DO NOT add CarDemo or DroneDemo scene into the scene hierarchy pane.

-Clicked on CarDemo or DroneDemo  got this error.
[image]

and Crashed how to solve this crash

rajat2004, 2020-04-27:
You can follow the steps mentioned in #2609 to try out PR #2328 

```
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git fetch origin pull/2328/head:PR/2328; git checkout PR/2328;
./build.sh; # or build.cmd for Windows
cd Unity
./build.sh; # or build.cmd for Windows
```

blacknaki, 2020-04-28:
# update if not already: git pull upstream master --rebase

how to do that step?

rajat2004, 2020-04-30:
These are commands to be run from the terminal, `git checkout master` changes the branch to be master
`git pull origin master --rebase` will apply the latest master branch changes. If you configured the current Airsim repo's remote to be `upstream`, then change `origin` to `upstream` in the command

rajat2004, 2020-05-01:
#2328 has been merged, so I think this issue can be closed

closed 2020-08-14 by hrnbot