microsoft/AirSim #2630: Running simulations with a time multiplier

vguillet, 2020-04-28:
OS: Windows
Python: 3.7

I am planning on using AirSim for reinforcement learning as part of a university project. Through reading the documentation, I saw that it was possible to run simulations without rendering, which provides a significant boost in performance, but I was wondering if it was possible to also run sped up simulations (at x2 speeds for example and preferably more). This would allow for much more iterations to be performed in much less time, and as a result significantly speed up training.

Is this already a feature and I missed the documentation? And if not is there plans for it?

Cheers!

rajat2004, 2020-04-28:
`ClockSpeed` might be what you're looking for - https://github.com/microsoft/AirSim/blob/master/docs/settings.md#clockspeed

vguillet, 2020-04-28:
Yep, thats exactly what I was looking for, thank you!

closed 2020-08-10 by jonyMarino