microsoft/AirSim #2631: Car in UnityDemo Project does not move using PythonClient scripts

sebastian-ruiz, 2020-04-28:
I am running PR #2328 to avoid the Unity CarDemo project from crashing on startup.

On Windows10 , running Unity 2019.2.17f1.
I am running the `UnityDemo` project.
I get 100s of logs in the unity console log:

```
Control Mode:  Keyboard Vehicle=PhysXCar
UnityEngine.Debug:Log(Object)
```
None of the python scripts in the `PythonClient/car` folder make the car in the Unity demo project move. I can however move the car using the WASD keys and the arrow keys.

When running a Python script, I get the following messages in Unity repeatedly:

```
Control Mode:  Keyboard Vehicle=PhysXCar
Control Mode:  API Vehicle=PhysXCar
Control Mode:  Keyboard Vehicle=PhysXCar
Control Mode:  API Vehicle=PhysXCar
...
```

When I try and run hello_car.py and running the Unity demo, the python script reports 'connected', however the car doesn't move. The same happens with the `drive_straight.py` script.

higho51420, 2020-05-08:
i meet the same problem. do you solve it ?

ruzicam, 2020-05-31:
Did someone solved it?

higho51420, 2020-06-04:
I didn't solve it. I don't use latest version, i use airsim 1.2.1,unity 2018.2.14.  no problem.  I have no idea to solve it.

weigangwen, 2020-09-10:
> I didn't solve it. I don't use latest version, i use airsim 1.2.1,unity 2018.2.14. no problem. I have no idea to solve it.

What version of vs do you use，please

PR #2937 (2021-05-24): [Unity] Fix Car API
Set API enabled so that it uses API controls rather than keyboard
Plus fix some warnings

Will be easier to review commit wise

Closes #2679, closes #2636, closes #2631

Some more cleanup needs to be done, `keyboard_controls_` is actually incorrect and doesn't contain the keypresses, that is actually been done [here](https://github.com/microsoft/AirSim/blob/master/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Vehicles/Car/Car.cs#L58-L71). It can be seen that the external controls are applied only when `isApiEnabled` is `true`, which is being added in this PR.
[`CarPawn`](http [... cut]
closed 2021-05-24 by zimmy87