microsoft/AirSim #2632: Incompatible with X7

randjamarko, 2020-04-28:
I have a taranis qx7 and i want to connect it to my pc so i can control the multirotor in airsim. I tried copying the binary file, but it just keeps telling me it's incompatible when i try to restore it. Oh yeah, qx7 has the same firmware as x9d.
Anyone knows how to resolve this?

randjamarko, 2020-05-20:
So I solved this by deleting the joystick in Windows and installing it again.

closed 2020-05-20 by randjamarko

zardosht, 2022-02-02:
@randjamarko Hi. Can you please explain more how you solved this problem? We also have the same issue: 
* Taranis QX7 Access
* Update firmware to the latest: 2.3.14 (2021-07-23)
When trying to load the model into the RC we get the message "Incompatible".

randjamarko, 2022-02-03:
As I said, what helped me was deleting the joystick from windows (in windows devices) and reinstalling it. Didn't use any provided binary file