microsoft/AirSim #2633: Visual_Studio issue.

johan606303, 2020-04-28:
Hi.

Before al was more or less ok. I worked on my project with Spyder to code on Python (Conda environment).
Then I tried to work with Jupyter Notebook. Then I got the following issue.

[image]

Now it's the case with VS2019. I uninstalled and re-installed, but issues remain.

Thanks for help.

rajat2004, 2020-04-29:
This is a known issue with Jupyter and `msgpack-rpc-python` having conflicting Tornado version dependency. Best way would be to create a virtual env with just the libraries needed

johan606303, 2020-04-29:
Thank you, it works well now.

rajat2004, 2020-04-29:
Please do close the issue if the problem has been solved

lovettchris, 2020-04-29:
perhaps we should put this in an FAQ somewhere.

closed 2020-04-29 by johan606303

rajat2004, 2020-04-30:
@lovettchris This info is there in the docs - https://microsoft.github.io/AirSim/apis/#typeerror-unsupported-operand-types-for-asyncioloop-and-float
Though the placement can be improved