microsoft/AirSim #2637: Unable to control multiple cars with Python API

InfraredLaser, 2020-04-29:
Hello,

I have an issue controlling multiple vehicles using Python APIs.

Here are my specs:
-UE 4.24.3, VS2019, Python 3.7, Windows 10

I looked through this documentation [https://microsoft.github.io/AirSim/multi_vehicle/](url) to pilot multiple cars as well as run a simple script in the image below. I can get multiple cars to spawn, however only the first vehicle responds to API commands.

Here is an image of my code and simulation environment:
[image]

Here is my `settings.json` file:
[image]
 
From what I can only assume, the simulator is able to identify multiple vehicles, but I am not quite sure why the second vehicle is not responding.

How am I able to use APIs to control multiple vehicles?

Thank you

InfraredLaser, 2020-05-04:
So I took a look in to the Issues within the repository and tried/noticed a few things:

- Issue #1008 as per @buribae (Stephen Jang) states that simply adding port numbers in the **settings.json** like my below sample code is ineffective at controlling two vehicles with Python API.
`{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
    "Car1": {
      "VehicleType": "PhysXCar",
      "ApiServerPort": 41451
    },
    "Car2": {
      "VehicleType": "PhysXCar",
      "X": -5,
      "Y": -4,
      "Z": 0,
      "ApiServerPort": 41452
    }
  }
}`
I thought that simply adding two SUV's by dragging and dropping their blueprints then running the simulation with the above **settings.json** would do the trick but it still does not work. I think you are only able to add drones by dragging and dropping their blueprints into their environments but not SUV's? I was also unable to find the file **swarmControl.py** from @saihv repository to see if the functionality in that file could be extracted to control multiple SUV's.  Most of these responses are a few years old and I am hoping there is an easier way to add and control vehicles with API's.

buribae, 2020-05-04:
@InfraredLaser Sharing in case you need [swarmControl.py](https://github.com/buribae/AirSim/blob/spawn_api/PythonClient/swarmControl.py)

buribae, 2020-05-04:
@InfraredLaser BTW, it seems like AirSim people came up with codes for cars and drones.
[car](https://github.com/microsoft/AirSim/blob/master/PythonClient/car/multi_agent_car.py)
[drone](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/multi_agent_drone.py)

InfraredLaser, 2020-05-04:
Thank you @buribae for the file PATH. I also took a look at the sample code for both car and drone above yet only the first vehicle is able to move with API's.

InfraredLaser, 2020-05-04:
Ahhh silly me, after running the car file above yet again both vehicles suddenly moved with API's! 
I took a look of the example code that I was running and compared it with the above code and it turns out that in my code I did not add the car name as a parameter in the `setCarControls` function, so only the first vehicle was responding to API commands.

This is the proper notation for setting the controls for a second vehicle and possible more. 
`client.setCarControls(car_controls_2, "Car2")` 

Now at least both of them can move! Thank you @buribae for the help!

closed 2020-05-04 by InfraredLaser