microsoft/AirSim #2638: ROS topic rates are very very low

BritskNguyen, 2020-04-30:
Hi all,

I am trying to run the first [AirSim tutorial](https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/#single-drone-with-monocular-and-depth-cameras-and-lidar) and is encountering very low topic rates.

My setup is as follows:
+Ubuntu 18.04
+Intel® Core™ i7-9750H CPU @ 2.60GHz × 12 
+GeForce RTX 2080 with Max-Q Design/PCIe/SSE2
+nvidia-driver-440
+Linux 5.3.0-51-generic
+ROS melodic

So after starting the _Blocks_ binary I launched the ros node and found that all topics have very low rates for all of the topics. Below is a screenshot of the terminal when I run `rostopic hz /airsim_node/drone_1/lidar/LidarCustom`:

[image]

The CPU usage is pretty normal, please see below:
[image]

I wonder where the issue is? Could it be because of the setup? Should I downgrade to Ubuntu 16.04 or use docker?

Thanks in advance!

rajat2004, 2020-04-30:
There's currently a bug in the release in Lidar where `DrawDebugPoints` if True will drop the FPS like anything. A fix (https://github.com/microsoft/AirSim/pull/2614) has been merged in master
Try setting DrawDebugPoints to False in the settings and see if it improves

BritskNguyen, 2020-04-30:
Thanks after setting the parameter to false I can get 5Hz of img, 25Hz of lidar and 30Hz of IMU. I think the issue can be closed. Will check (#2369) for customizing the rate.

closed 2020-04-30 by BritskNguyen