microsoft/AirSim #2640: Cannot find airsim module (AirSim)

AngelTang190, 2020-04-30:
I tried to make the quadcopter in AirSim fly. I used Anaconda Prompt (anaconda3) and I had my unreal engine opened with the quadcopter simualtion running. I went to the directory AirSim\PythonClient\multirotor and ran python hello_drone.py and it gave me this error:

(base) C:\code\AirSim\PythonClient\multirotor>python hello_drone.py
Traceback (most recent call last):
  File "hello_drone.py", line 2, in <module>
    import airsim
ModuleNotFoundError: No module named 'airsim'

Can someone please help? Thanks!

rajat2004, 2020-04-30:
Did you run `pip install airsim`?

AngelTang190, 2020-04-30:
> Did you run `pip install airsim`?
I've previously build airsim using developer command prompt for VS 2019. Should run pip install airsim again?

rajat2004, 2020-04-30:
That builds the AirSim module, you need to install the `airsim` python package to use the Apis

AngelTang190, 2020-04-30:
After I run pip install airsim and run python hello_drone.py again, nothing happens and the Anaconda is not responding as well.

rajat2004, 2020-04-30:
Simulation might not be running
Else if there are any errors or crashes, please post the entire steps to reproduce the issue

AngelTang190, 2020-04-30:
First, I started unreal engine and opened my project, Then, I hit play and chose quadcopter as my vehicle. Then, I opened Anaconda prompt and I typed in 

cd\code\AirSim\PythonClient\multirotor\

After that, I typed in 

python hello_drone.py

These are all the steps.

There are no errors shown on Anaconda prompt.

AngelTang190, 2020-04-30:
[image]

AngelTang190, 2020-04-30:
[image]

rajat2004, 2020-04-30:
Seems pretty strange, I can't reproduce though I am on Linux. And there would be a lot more issues if something like this is happening
Try running the Blocks env and see if it's occurring in that as well

AngelTang190, 2020-04-30:
Do you mean Blocks.uproject?

closed 2020-05-14 by AngelTang190