microsoft/AirSim #2643: How to update custom Unity environment to latest version of AirSim?

mateusk, 2020-04-30:
Hi,

First of all, thanks for maintaining this great open source project!

I have a custom Unity environment and I wonder what would be the recommended steps to update AirSim to the latest version. I tried simply to replace the "AirsimWrapper.dll" but I ran into similar issues as in [#2629](https://github.com/microsoft/AirSim/issues/2629). 

I'm on Windows 10 1909, using Unity 2019.3.12f1.

Thanks!

rajat2004, 2020-04-30:
Updating your local repo with the PR fixes as mentioned in the issue, rebuilding and then following the steps for custom Unity env should work

mateusk, 2020-05-08:
Ok, thanks, I believe that worked. Is it absolutely needed to copy `/ProjectSettings` as well?

jonyMarino, 2020-08-14:
@mateusk did you try not copying them? I would like to close this issue

closed 2020-08-28 by jonyMarino