microsoft/AirSim #2646: Setup Blocks environment for AirSim not working (Windows)

David-Sola, 2020-05-01:
Hi all,

I just installed everything as described in the docs an try to setup the blocks environment as described here: https://microsoft.github.io/AirSim/unreal_blocks/ 
I have build everything and have the following fodlers already available:
AirSim/Unreal/Plugins/AirSim --> exists with content
Double clicked on update_from_git
.sln files exists and also Blocks/Plugins/AirSim with content

When I know open VS2019 and try to run it I get the following error(s)

Fehler	MSB3075	Der Befehl "D:\Unreal\UE_4.24\Engine\Build\BatchFiles\Build.bat BlocksEditor Win64 DebugGame -Project="D:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" wurde mit dem Code 5 beendet. **Stellen Sie sicher, dass Sie zur Ausführung dieses Befehls berechtigt sind.**	Blocks	D:\Visual_Studio_Gaming\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	44	

It says: Please be sure to have the rights to execute this command

Above this error I have multiple erros which say:
Fehler	LNK2001	Nicht aufgel÷stes externes Symbol "__GSHandlerCheck_EH4".	Blocks	D:\AirSim\Unreal\Environments\Blocks\Intermediate\ProjectFiles\MavLinkCom.lib(MavLinkMessageBase.obj)	1	

Could you help me with this issue?

UE version: 4.24
I also just recognized, that even though I have opened the sln file with VS2019 I can see it it displays VS2017. Could this be a hint to an issue?
[image]

Thank you very much,
David

johan606303, 2020-05-01:
Had you not open this Blocks as Administrator ? I'm a beginer with airsim too. But Blocks is the only project for the moment that I have to open as Administrator.

David-Sola, 2020-05-01:
I started VS2019 as an Admin but it didn´t resolve the issue

johan606303, 2020-05-02:
You can perhaps to open `cmd Prompt` as Administrator (I made it with x64 Native Tool Command Prompt for VS 2019) and try the following commands : 
1st go to directory of airsim (where you cloned AirSim), then :

`git checkout master`
`git fetch origin`
`git rebase origin/master`

If this last command returns : 
[image]
run the command `git stash`
and then  `git rebase origin/master`

After that you will need to build AirSim `build.cmd`, go to Blocks with the terminal and run `update_from_git.bat`.

closed 2020-08-10 by jonyMarino