microsoft/AirSim #2647: PX4 in SITL

johan606303, 2020-05-01:
Hi.
I am trying to set PX4 in SITL mode.
The commands : `git checkout v1.10.1` and `git checkout v1.10.2` .
So I tried with `git checkout v1.11.0-beta1` and I arrived to : 
[image]

So after I modified `Settings.json` : 

[image]

But when I run a project, `run-console.bat` returns the following issues : 
[image]
 And on UE4 I have the following messages : 
[image]
What do I have to do else to control the drone using Python ? 

Thanks for your help.

rajat2004, 2020-05-01:
Are you running the latest AirSim master? If not, then try with that, several issues with PX4 have been fixed very recently

johan606303, 2020-05-01:
Do you mean run the command  `git rebase origin/master` ?

rajat2004, 2020-05-01:
git fetch origin
git rebase origin/master

johan606303, 2020-05-01:
Well I ran those commands, it returned to me :
[image]

So the issue is a bit different. But I forgot to mention that when I run a project, when UE4 appear and I press `Play`, VS2019 says that i meant `breakpoints`
[image]
[image]

rajat2004, 2020-05-01:
You might be running into the problem discussed in https://github.com/microsoft/AirSim/issues/2583
@lovettchris would have a better idea
Just to confirm, you're not running in HITL and only SITL correct? Also, are you using any joystick?

johan606303, 2020-05-01:
It's for SITL, and I don't use joystics, I don't have any remote controle. I want to controle just through executing codes, at least for now.

rajat2004, 2020-05-01:
Just to check if the current latest master is being used, could you put the last commit id
`git log` should give the list of commits

Also, did you run `build.cmd` and then `update_from_git.bat` to update the plugin

lovettchris, 2020-05-01:
The heap corruption in AirSim was fixed with this PR #2634, can you make sure you have those bits installed ?  I built this fix into these binaries, and I've tested those binaries with PX4 v1.11.0-beta1:

- [AirSim Neighborhood binary (1.3gb)](https://drive.google.com/file/d/1DIleir72Jzfx3oU2i8fEHNBI-37UfXpA/view?usp=sharing) 
- [AirSim Blocks binary (153mb)](https://drive.google.com/file/d/1OMvLO2LcNI5KxYovYKTWj9XGGXHYXvNC/view?usp=sharing) 

I also see in your screen shot where it talks about lighting needs to be rebuilt, so I wonder if Unreal is still compiling shaders in the background, which could create some timing problems...  My pre-compiled binaries will not have that problem.  This shows a python script flying the above AirSim Neighborhood using PX4 SITL v1.11.0-beta1:

[image]

johan606303, 2020-05-02:
When I run `git log` I have : 

[image]

johan606303, 2020-05-02:
Well I ran commands `build.cmd`, and `update_from_git.bat`. Then when I ran the project Blocks, on PX4 `run-console.bat` I have : 
[image]

and on UE4 : 

[image]
 Then if I try to run api `armDisarm(True)`
[image]

and on UE4 I have : 

[image]

bys1123, 2020-05-02:
@lovettchris Thanks, I'll test these binary.

johan606303, 2020-05-04:
@bys1123 Hi, does it work for you ?

bys1123, 2020-05-04:
https://github.com/microsoft/AirSim/pull/2649
@johan606303 Not yet, here is new PR and new binaries, might works.

johan606303, 2020-05-04:
Ok, thank you for the link.

johan606303, 2020-05-04:
@lovettchris Ok. I tried to run with Blocks or Neighborhoood, that @bys1123 sent me. But I still have those issues, and I con't understand why. On `run-console.bat` : 

[image]

and on VS 2019 when I try to run some API python commands, it sends me an exception : 

[image]

Something is wrong with the GPS, but I defined them in `settings.json` as on tutorial : 

[image]

Or running PX4 in SITL rearly requires to connect and exterior device ?

bys1123, 2020-05-16:
[image]
@johan606303 I failed too.

johan606303, 2020-05-20:
@bys1123 Ok. If I find something, I will tell you.

DanielGilo, 2020-06-01:
I'm facing similar issues, are there any updates?

lovettchris, 2020-08-27:
Note that there are 2 PX4 messages you should be looking for:
```
INFO  [ecl/EKF] 3859668: EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] 8418081: EKF commencing GPS fusion
```
The first one does not mean it is ready to fly.  The second one means AirSIm can fly, and in AirSim you should see an additional message saying "Got a home GPS location".  AirSim cannot fly until this happens.  Sometimes this second message takes a while depending on PX4.  You can also try setting the following using the PX4 console window just in case:

```
param set LPE_LAT 47.641468
param set LPE_LON -122.140165
param set COM_OBL_ACT 1
param set NAV_RCL_ACT 0
param set NAV_DLL_ACT 0
```

just paste that block into PX4 console window and hit enter.  The AirSim settings.json file should be doing this but just in case it isn't, succeeding in passing those settings to PX4, doing it manually should also help.

jonyMarino, 2020-11-06:
Is this related to the issues reported in #2477?

jonyMarino, 2021-06-01:
Please try our [new version](https://github.com/microsoft/AirSim/releases) which has many updates for px4.
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-06-01 by jonyMarino