microsoft/AirSim #2648: Startup of Car- and Drone-mode fails on Unity

SirRobbe, 2020-05-01:
In order to reproduce the problem:

- Install AirSim for Unity like described here: https://microsoft.github.io/AirSim/Unity/
- Open the Unity Project and start the SimModeSelector Scene
- Pick either Car or Drone Mode

Loading the according scenes fails and following error message is reported for car mode:

Cannot read settings file  G:\Engines\Unity\Versions\2019.3.3f1\Editor\settings.json Vehicle=PhysXCar
UnityEngine.Debug:LogError(Object)
AirSimUnity.Vehicle:PrintLogMessage(String, String, String, Int32) (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:293)
AirSimUnity.VehicleCompanion:PrintLogMessage(String, String, String, Int32) (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:199)

blacknaki, 2020-05-03:
I get the same error. Any solutions?

TomSeestern, 2020-05-04:
please share your settings.json

SirRobbe, 2020-05-05:
I could not find a settings.json. Where should this file be located?  Also could it be that it tries to load the file from the wrong place, since the path reported in the error is the path to the installed Unity version, however there aren´t any files specific to one project in this directory.

AngelTang190, 2020-05-09:
> I could not find a settings.json. Where should this file be located? Also could it be that it tries to load the file from the wrong place, since the path reported in the error is the path to the installed Unity version, however there aren´t any files specific to one project in this directory.

It's under Documents -> AirSim

AngelTang190, 2020-05-09:
> In order to reproduce the problem:
>
> * Install AirSim for Unity like described here: https://microsoft.github.io/AirSim/Unity/
> * Open the Unity Project and start the SimModeSelector Scene
> * Pick [...]

I'm also facing the same problem as well

AngelTang190, 2020-05-09:
[image]

There are only two lines of code here, but I didn't change anything.

SirRobbe, 2020-05-09:
> It's under Documents -> AirSim

Thank you! My settings look like this:

`{
 "SimMode" : "", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}
`

AngelTang190, 2020-05-09:
> > It's under Documents -> AirSim
> 
> Thank you! My settings look like this:
> 
> `{ "SimMode" : "", "SettingsVersion" : 1.2, "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md" } `

Same as yours, just the lines are inverted.

blacknaki, 2020-05-10:
{
 "SimMode" : "", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}

Gastastrophe, 2021-05-26:
For me, when I opened my settings.json file in Visual Studio, it said that the line ending formatting was mismatched and I was provided a list of options to change it to. I selected Windows, since I was working on a Windows machine. As a note, I also did the recommended fix above by placing a copy of my settings.json file in "Unity\Versions\2019.3.3f1\Editor" where the error message said it was searching. This fixed the issue for me.

Ibrahim2595, 2022-05-06:
Any solution to this so far? I am getting the same issue still and I have this as my json file.

{
 "SimMode" : "", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}

I also opened the file with visual studio and accepted the recommendations to fix for windows. But error is still there.