microsoft/AirSim #2653: How to run inference on pre-trained YOLO model (Object detection)

thejeshk, 2020-05-03:
Hi, 

I have created my own custom environment and also my own custom vehicle in Unreal and built airsim. Now Airsim is plugged in my custom environment

Project Pipe-line: Create environment and Vehicle --> Object detection (for custom class e.g. Traffic cone) --> Path planning (RRT*) --> PID controller --> Implement in **Real vehicle** 

Now I am stuck at object detection part. I have manually collected the traffic cone images from my environment and trained with YOLO model in cloud. So, now I have trained weights and I need to know if there is any possible way to run inference on pre-trained YOLO model. 

What I need is --> When my vehicle is running in Airsim environment, when it sees the traffic cone it has to detect and put bounding box over it. 

Note: I am doing this in Airsim and leteron if this works then we can implement in real vehicle. 

Please let me know how can run inference on pre-trained YOLO model 

Thank you :)

rajat2004, 2020-05-03:
I'm not sure if I understand this correctly, but can't you just fetch the images using `simGetImages` API, and then pass them to your model?
As to the bounding box, I think you mean drawing that on the image which can be done by OpenCV or something else, or do you want to draw the bounding box on the object in the env itself? In the latter case, the plotting APIs might be of use

thejeshk, 2020-05-04:
Hi @rajat2004 

Thank you for quick response. 

I want to draw bounding boxes over object (traffic_cone) in an environment. what do you mean by plot API's ??

thejeshk, 2020-05-04:
Plotting apis*

thejeshk, 2020-05-19:
> Hi,
>
> I have created my own custom environment and also my own custom vehicle in Unreal and built airsim. Now Airsim is plugged in my custom environment
>
> Project Pipe-line: Create environment and [...]

closed 2020-05-19 by thejeshk

postureai, 2020-07-18:
> Hi,
>
> I have created my own custom environment and also my own custom vehicle in Unreal and built airsim. Now Airsim is plugged in my custom environment
>
> Project Pipe-line: Create environment and [...]

Hii, Did you able to do this? Could you please let me know how?