microsoft/AirSim #2662: Control API's for drone in Python

rovdrcru, 2020-05-05:
Hello!

I would like to make a drone fly autonomously based on simulated signal information. Therefore I'm searching for (Python) API's to control the drones movement. 

Below is a .hpp file that I already found and that I could use, but the problem is that I'm a layman when it comes to C++ 
https://github.com/microsoft/AirSim/blob/master/AirLib/include/common/PidController.hpp

If there are any made in Python, where would I be able to find them?
If there are good c++ ones, they are as well welcome, then I could consider switching to c++.

Sidenote: I'm pretty new to programming and the use of API's. 

I'm using:
* python 3.8
* unreal engine 4.24.3

ndalton12, 2020-05-05:
AirSim has a Python API you can [use](https://github.com/microsoft/AirSim/tree/master/PythonClient). It allows you to do various things like control velocity or set the position of the drone. It should be fairly easy to implement your own PID controller from there.

rovdrcru, 2020-05-05:
Thank you for your answer. 
Do you mean that I should use a self build PID controller in python and use inputs and outputs of airsim retrieved through airsim API's?

ndalton12, 2020-05-05:
Yes, exactly.  For example, you could use `moveByVelocityAsync()` and a PID controller to set the velocities appropriately based on a sensor signal. Although, `moveByVelocityAsync()` probably uses a PID controller itself under the hood based on motor encoders/PWM or something, but I am not sure.

closed 2020-08-10 by jonyMarino