microsoft/AirSim #2663: how to control airsim quadrotor through Python

ashish210111, 2020-05-05:
I have successfully installed Unreal Engine 4.24, and Airsim is also connected successfully.  Now I am able to drive the car in Landscape Mountains project using a keyboard.  Now how to control the quadrotor using python scripts?

johan606303, 2020-05-06:
You can for example use VS2019 to code in Python, or for example Spyder (but in my case I always had to re-open Spyder after each single run). In VS2019, You need to go to `Tools`-->`Python`-->`Python environment`, and install modules `msgpack-rpc-python` and `airsim`. You can as well create an environment and istall those modules. You can then either open an interactive window to send commands one by one, or create a new VS2019 Python project, and from there run commands.

ashish210111, 2020-05-06:
Thanks for your reply I have installed anaconda and using anaconda installed msgpack
now how to choose car and navigate to PythonClient\car\ folder and run. 
[image]

closed 2020-05-08 by ashish210111