microsoft/AirSim #2664: Copy of Landscape Mountain Environment Uproject

ndalton12, 2020-05-05:
Hi, I am looking to see if anyone can point me to a download for the Landscape Mountain environment uproject (not the binary). I want to add in some custom drones, but I can't find it available on the Epic Games launcher anymore, so I'm wondering if there is a download elsewhere.

Thanks!

rajat2004, 2020-05-11:
Yes, it seems to have disappeared. BTW, while searching, I found this, might be interesting ti people here, especially those who want City like env - https://www.unrealengine.com/marketplace/en-US/product/modern-city-downtown-with-interiors-megapack

jonyMarino, 2020-09-23:
Now it is in the unreal section in the learn tab under legacy. I just downloaded it

closed 2020-09-23 by jonyMarino