microsoft/AirSim #2665: Why picture pixels cannot be converted to floating-point numbers

LLalang, 2020-05-06:
responses = client.simGetImages([airsim.ImageRequest("0", airsim.ImageType.DepthPerspective, True, True)])
current_state = transform_input(responses)

def transform_input(responses):

    img1d = np.array(responses[0].image_data_float, dtype=np.float)

    img1d = 255/np.maximum(np.ones(img1d.size), img1d)

    img2d = np.reshape(img1d, (responses[0].height, responses[0].width))

    from PIL import Image
    image = Image.fromarray(img2d)
    im_final = np.array(image.resize((84, 84)).convert('L')) 

    return im_final
[image]

LLalang, 2020-05-06:
Has anyone ever encountered such a problem？

rallen10, 2020-07-22:
I have the same problem. `image_data_float` is `[0.0]` even though `image_data_uint8` has large amounts of data. I am using `pixels_as_float=True`. Did you ever find a solution?

sandilyasg, 2023-11-06:
@LLalang were you able to resolve this issue? I am using Unity and image_data_float always returns an empty image. However, when I view the depth, semantic segmentation, and RGB images directly in Unity, they appear to be working fine.
[image]