microsoft/AirSim #2666: implementing non-CVMode keyboard controller for quadcopter

avivazran, 2020-05-06:
Any ideas regarding the best way to implement keyboard control for the quadcopter NOT in CVMode?
i've managed to use python to enable keyboard interactions to read data but I lack the knowledge to create the code for a PID based movement controller.
I've thought about implementing it via inherited drone class directly to unreal engine but this requires me to take a course in unreal engine development.

any ideas?

TheFloe1995, 2020-05-11:
I used the high level API (i.e. `moveByVelocityAsync()`) for that purpose. Simply map keyboard keys to specific directional vectors in the drone's coordinate frame, combine them and transform the resulting velocity vector into the global frame. Use the `yaw_mode` parameter to make the drone turn simultaneously.

avivazran, 2020-05-11:
@TheFloe1995 
First, thank you for the reply.
I've tried this method but i it results in a very non-flowing and shaky movement.
is there a chance you could share the code of your implementation?

TheFloe1995, 2020-05-12:
Sure, have a look at [this Gist](https://gist.github.com/TheFloe1995/f67278952bd8eb5ff59d463f9dc48d53). It's not perfect, you probably won't win a race with it (tends to drift a lot).

Essentially you need to call the async commands in an overlapping fashion without joining them.