microsoft/AirSim #2670: Cannot build Unreal Project from Visual Studio 2019

coolboyhy1607, 2020-05-07:
I do the step as below but cannot build Block.sln by VS2019.
1. run build.cmd in AirSim directory.
2. Copied AirSim/Unreal/Plugin folder to Unreal\Environments\Blocks folder.
3. Run  update_from_git batch file.
4. Open Blocks.sln file, select Blocks as start project, select DebugGame_Editor with Win64.
5. Hit F5
Then the error occurred. The build log as below.
===========
[Build log.txt](https://github.com/microsoft/AirSim/files/4590168/Build.log.txt)
===========
The log says that "fatal error LNK1120 2 unresolved externals".

I have tried to make my original Unreal Project by following [step-by-step guide](https://microsoft.github.io/AirSim/unreal_custenv/) but it was not changed.

I have tried to open Blocks.uproject by Unreal Engine and the popup dialog showed like the picture below.
[image]

I use Window10, Unreal Engine 4.24.3, AirSim 1.3.1, VS Professional 2017, and VS Community 2019.

Please tell me what I can do.

rajat2004, 2020-05-07:
From the log, it can be seen that it's using VS 2017 to build the project. Either uninstall that, or you can try https://github.com/microsoft/AirSim/pull/2526#issuecomment-611274394 for generating VS 2019 sln

coolboyhy1607, 2020-05-07:
> From the log, it can be seen that it's using VS 2017 to build the project. Either uninstall that, or you can try [#2526 (comment)](https://github.com/microsoft/AirSim/pull/2526#issuecomment-611274394) for generating VS 2019 sln

Thank you. The problem was solved.

closed 2020-05-07 by coolboyhy1607