microsoft/AirSim #2677: Sometimes drone shakes instead of moving

AllenIverSunn, 2020-05-08:
Hi,
When I try to run reinforcement learning with tensorflow and use Airsim as the simulator, sometimes in the neighborhood environment provided by Airsim, the drone shakes instead of taking an action. More detailed,  the drone move by 3 unit each time an action is generated and sometimes it go by 3 units nicely while sometimes it just shake towards the direction without moving. Does anyone else have the same issue? I'd appreciate it if someone can enlighten me. THANKS!

AllenIverSunn, 2020-05-12:
I changed my deep learning code from Tensorflow to Pytorch and everything becomes fine. Is it because of this?

madratman, 2020-08-13:
It is unclear if your issue was due to airsim APIs or in your own code. Moving from tf to pytorch doesn't seem as informative.
It would be great if you can share which drone "moveBy" API you were using and if you ended up changing the amount of duration in your calls / or sleep b/w your calls to the API.

jonyMarino, 2020-08-26:
closed due to incompleteness. Please feel free to open a new issue with the requested information

closed 2020-08-26 by jonyMarino