microsoft/AirSim #2684: Importing UE4 Model into AirSim

caesar84, 2020-05-11:
I have followed the steps of @Chris Lovett in this [https://www.youtube.com/watch?v=1oY8Qu5maQQ&ab_channel=ChrisLovett](video1) that illustrate how to import a UE model into AirSim.
My issue is what if I have my own model (project) that has a different folder structure and content from the free to download models available in the UE library and used in the tutorial video above? 
to explain: I have a 3D max model that has been converted to UE project  Using DataSmith plugin according to this video tutorial [https://www.youtube.com/watch?v=1GH7QDYQXeI&ab_channel=LearnArchViz](video 2)  and the process went ok.
However, I have faced the following problem where video 2 doesn't show how to save the imported model in a format that matches the ''LandscapeMountains'' project format so I can follow video 1 from that point onwards to use it in AirSim. To be more specific :
project folder of ''LandscapeMountains'' contains many essential elements such as the DerivedDataCache, config, and Content.
the converted model using DataSmith firstly doesn't have DerivedDataCache folder. In addition, the Imported model Config files are by no means close to ''LandscapeMountains'' config files. Moreover, the converted model contents folder contains Developers, Collections, and exported_model folders plus a .umap file of the exported model. whereas ''LandscapeMountains'' content folder has only two folders: Maps and Assets folders. all these details are shown in images1,2,3 and 4
[image]
[image]
[image]
[image]
It would be highly appreciated if anyone can help with this issue in any shape or form considering 
 how easy is it now to import  3d max models into UE with the great help of the powerful Datasmith plugin. 
this could have a massive impact on opening up a wide range of solutions for model selections or design for any Drone ( My current work) or Car AI applications using AirSim.

finally, I have posted my the same problem on different forums before such as UnrealEngine community forum, StackOverflow, StackExchange/Game development) but no luck so far. I think Either I didn't explain it properly or its too complicated for them. Therefore, I hope I made it clear here especially the final goal is more familiar with AirSim guys.

thanks in advanced

rajat2004, 2020-05-11:
From what I know, I don't think the internal project structure as such matters that much as long as UE is able to run the environment. If the project is working with UE without AirSim, then to add AirSim, you just need to copy the Plugins folder to the project directory, modify the .uproject file to include AirSim as a plugin and it should work following the steps in the documentation

Also, the DerivedDataCache folder is probably built by UE when it compiles the shaders or other stuff, it's not there in the Blocks env as well
You are probably more experienced in working with UE, these are just my 2 cents

ndalton12, 2020-05-11:
Seconding what @rajat2004 said, you only need to copy the content folders in for UE to recognize them; it shouldn't make a difference if the other folders are different.

Personally, I think the easiest way to get a custom model working is to import your meshes, then make a copy of `BP_FlyingPawn.uasset` and modify it so that it uses your new meshes instead. This way you don't have to modify the code/blueprint graphs and the rotors will spin already.

closed 2020-07-29 by madratman