microsoft/AirSim #2688: Can't control the drone gimbals

agentfeux, 2020-05-11:
Hi,
I am using Airsim with ArduPilot Copter and QGroundcontrol ,and I'm trying to move the camera with the gimbals but its doesn't work.

**This is my Setting.json:**
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
   },
   "CameraDefaults": {
    "Gimbal": {
      "Stabilization": 1,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
    },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "RC": {
             "RemoteControlID": 0
            },
	  "SubWindows": [
	    {"WindowID": 0, "ImageType": 0, "CameraName": "3", "Visible": true},
	    {"WindowID": 1, "ImageType": 3, "CameraName": "0", "Visible": true},
	    {"WindowID": 2, "ImageType": 6, "CameraName": "4", "Visible": true}
	  ],
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002
    }
  }
}

**This is my QGroundcontrol config:**
[image]

**When i try to move my gimbal Mavlink get this:**
`Got MAVLink msg: COMMAND_ACK {command : 205, result : 0}`

Thanks you for helping me with this issue

rajat2004, 2020-05-12:
How are you trying to control the gimbal? Are you using the AirSim APIs or dronekit or something else? Try using [`simSetCameraOrientation`](https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#camera-apis)

agentfeux, 2020-05-12:
I was originally trying to control the gimbal with my Xbox one controller via Mavlink. Does Airsim support a gimbal controlled by Mavlink?

rajat2004, 2020-05-12:
AFAIK, it doesn't support Mavlink controlled gimbal. Any contributions would definitely be welcome!

agentfeux, 2020-05-12:
Thanks the information ! i know what to do now ;)

closed 2020-05-12 by agentfeux