microsoft/AirSim #2689: Compilation error on linux

GimmyUE4, 2020-05-12:
Hello,
I am currently trying to install UE4 but when I try to compile with make, I get an error:
`
ERROR: Cannot use an old toolchain (missing ToolchainVersion.txt file, assuming version earlier than v11)
`
I found only one thread describing this issue:
https://github.com/microsoft/AirSim/issues/2456

it might be same problem although they've never posted solution. Only bit of information I get is that I could have problem with proxy setting. And I have no clue what might be wrong there.

My system: Kernel: 5.4.39-1-MANJARO x86_64

Here's whole procedure:
```
[bartek@bartek-pc UnrealEngine-release]$ ./Setup.sh               
Fixing inconsistent case in filenames.
Setting up Mono
Checking dependencies...
The following file(s) have been modified:
  Engine/Binaries/ThirdParty/Mono/Linux/etc/mono/registry/last-btime
Would you like to overwrite your changes (y/n)? y
Updating dependencies: 100% (1/1), 0.6 MiB cached, done.
Result: 0
Register the engine installation...
Installing a bundled clang toolchain
Toolchain already installed skipping.

Target arch set to: x86_64-unknown-linux-gnu
Building ThirdParty libraries

If you don't see SUCCESS message in the end, then building did not finish properly.
In that case, take a look into /home/bartek/Pobrane/UnrealEngine-release/Engine/Build/BatchFiles/Linux/BuildThirdParty.log for details.

No third party libs needed to be built locally

**********  SUCCESS ****************
Setup successful.
[bartek@bartek-pc UnrealEngine-release]$ ./GenerateProjectFiles.sh
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
        updated UE4Printers.py
        found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Writing project files... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Writing project files... 100%
[bartek@bartek-pc UnrealEngine-release]$ make
bash "/media/Dysk_HDD/Pobrane/UnrealEngine-release/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
Fixing inconsistent case in filenames.
Setting up Mono
Building CrashReportClient...
Using 'git status' to determine working set for adaptive non-unity build (/media/Dysk_HDD/Pobrane/UnrealEngine-release).
Creating makefile for CrashReportClient (no existing makefile)
Creating makefile for UnrealHeaderTool (no existing makefile)
------- Build details --------
Using toolchain located at '/media/Dysk_HDD/Pobrane/UnrealEngine-release/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v16_clang-9.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/media/Dysk_HDD/Pobrane/UnrealEngine-release/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v16_clang-9.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '9.0.1' (string), 9 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /media/Dysk_HDD/Pobrane/UnrealEngine-release/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v16_clang-9.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Cannot use an old toolchain (missing ToolchainVersion.txt file, assuming version earlier than v11)
make: *** [Makefile:302: CrashReportClient-Linux-Shipping] Błąd 6
[bartek@bartek-pc UnrealEngine-release]$ 
```

rajat2004, 2020-05-12:
If your earlier `Setup.sh` completed correctly without any errors, which might not be the case going by the related issue, you could try something mentioned in https://forums.unrealengine.com/development-discussion/engine-source-github/1687144-missing-toolchainversion-txt-linux-ubuntu-19-04
Please comment on the linked issue to ask about the proxy settings

GimmyUE4, 2020-05-12:
Thanks it worked! For anyone with same problem here's solution. It's stupidly easy:
create ToolchainVersion.txt file inside:
`/home/bartek/Pobrane/UnrealEngine-release/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v16_clang-9.0.1-centos7/`
and write in the file: name of the folder in my case: "v16_clang-9.0.1-centos7"

closed 2020-05-12 by GimmyUE4

radhanathdas, 2020-09-05:
> Thanks it worked! For anyone with same problem here's solution. It's stupidly easy:
> create ToolchainVersion.txt file inside: [...]

It works for me too. :) Thank you!