microsoft/AirSim #2692: Recommended AirSim and PX4 versions for API control

DanielGilo, 2020-05-13:
Hi,
I'm working on an AirSim project on Windows, trying to use PX4 SITL while controlling the drone via Python API. I've faced many issues so I wanted to know what are the recommended versions of AirSim, PX4, and Unreal for the use I've described?

The issues I faced (AirSim version 1.2.2, Unreal 4.18.3/4.24.4):
Initially (with the default flight controller) the drone was shaking when flying over 50m, so I tried using PX4. Tried using v1.9.2 but had connection issues (PX4 expected TCP connection while AirSim was using UDP). The solution presented in https://discuss.px4.io/t/airsim-base-on-px4-v1-9-2/13696 didn't work (got segmentation faults when launching PX4). So I tried v1.8.2 where there weren't connection issues and I was able to control the drone via API, but it didn't exactly do as instructed (I'm trying to move the drone between a few positions, it gets to the first position and then starts moving strangely towards the second position, then landing early before visiting all the positions).  

Really all I want to do is  to fly with no issues between positions via API...  I'll be grateful for any help or recommendations.

rajat2004, 2020-05-13:
v1.2.2 is a very old release, and AFAIK, it supports PX4 1.8.2 which is also very old
You could try out the latest release, and the PX4 version recommended in the docs.
However, there have been fixes gone into master since the release, and there's another PR open - https://github.com/microsoft/AirSim/pull/2649 to fix some more
My suggestion would be to use the binaries linked in the PR with the latest PX4 release or master, whichever you prefer, and see if things work or not
Other suggestion would be to see if you can build AirSim from source. That'll allow you to run the latest master with the latest fixes, and any PRs as well, without waiting for a release which can take time

jonyMarino, 2020-08-13:
airsim's last release and current master support v1.10.1 
please see this document: https://microsoft.github.io/AirSim/px4_sitl/

closed 2020-08-13 by jonyMarino